KLayout/klayout #1981: PolygonsWithProperties branch ("devel")

martinjankoehler, 2025-01-27:
## Issues observed while testing the devel branch

- [ ] Reused LVSDB: dbShape ERROR

martinjankoehler, 2025-01-27:
### Reused LVSDB: dbShape ERROR

```
./kpex.sh --pdk sky130A --gds testdata/designs/sky130A/test_patterns/single_plate_100um_x_100um_li1_over_substrate.gds.gz --out_dir output_sky130A --2.5D y
───────────────────────────────────────────────────────────────── Command line arguments ─────────────────────────────────────────────────────────────────
/Users/martin/Source/klayout-pex/klayout_pex/__main__.py --pdk sky130A --gds
testdata/designs/sky130A/test_patterns/single_plate_100um_x_100um_li1_over_substrate.gds.gz --out_dir output_sky130A --2.5D y
────────────────────────────────────────────────────────────────────── Input Layout ──────────────────────────────────────────────────────────────────────
[16:53:36] INFO     GDS input file passed, running in LVS mode                                                                             kpex_cli.py:291
[16:53:36] INFO     No explicit top cell specified, using top cell 'single_plate_100um_x_100um_li1_over_substrate'                         kpex_cli.py:327
[16:53:36] INFO     LVS input schematic not specified (argument --schematic), using dummy schematic                                        kpex_cli.py:374
───────────────────────────────────────────────────────────────────── Prepare LVSDB ──────────────────────────────────────────────────────────────────────
[16:53:36] WARNING  Cache hit: Reusing cached LVSDB                                                                                        kpex_cli.py:712
output_sky130A/.kpex_cache/sky130A/Users/martin/Source/klayout-pex/testdata/designs/sky130A/test_patterns/single_plate_100um_x_100um_li1_over_substrate.gd
s.gz/single_plate_100um_x_100um_li1_over_substrate.lvsdb.gz
[16:53:36] WARNING  Unable to find info about extracted LVS layer 'l9'                                                              lvsdb_extractor.py:186
──────────────────────────────────────────────────────────── Non-empty layers in LVS database ────────────────────────────────────────────────────────────
[16:53:36] INFO     (67, 20) -> (li_con)                                                                                                   kpex_cli.py:767
────────────────────────────────────────────────────────────────── kpex/2.5D PEX Engine ──────────────────────────────────────────────────────────────────
[16:53:36] INFO     (Overlap): li1(PLATE)-VSUBS(VSUBS): Unshielded area: 10000.0 µm^2, cap: 369.9 fF                                      extractor.py:295
ERROR: src/db/db/dbShape.h,1218,m_type == Polygon
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/Users/martin/Source/klayout-pex/klayout_pex/__main__.py", line 32, in <module>
    main()
  File "/Users/martin/Source/klayout-pex/klayout_pex/__main__.py", line 29, in main
    cli.main(sys.argv)
  File "/Users/martin/Source/klayout-pex/klayout_pex/kpex_cli.py", line 826, in main
    self._rcx25_extraction_results = self.run_kpex_2_5d_engine(  # NOTE: store for test case
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/martin/Source/klayout-pex/klayout_pex/kpex_cli.py", line 608, in run_kpex_2_5d_engine
    extraction_results = extractor.extract()
                         ^^^^^^^^^^^^^^^^^^^
  File "/Users/martin/Source/klayout-pex/klayout_pex/rcx25/extractor.py", line 95, in extract
    cell_extraction_result = self.extract_cell(cell_name=cell_name, report=report)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/martin/Source/klayout-pex/klayout_pex/rcx25/extractor.py", line 738, in extract_cell
    shapes_inside_net.complex_op(node)
RuntimeError: Internal error: src/db/db/dbShape.h:1218 m_type == Polygon was not true in Region.complex_op
```

martinjankoehler, 2025-01-28:
[image]

klayoutmatthias, 2025-01-29:
I have pushed a fix. 

However, some more features are needed to make the code work as it is right now. Specifically, as Region does not interact well with PolygonWithProperties currently. I have added another commit to provide the basic features. For me, the example you mentioned now works. There is a little work left however.

BR,

Matthias

martinjankoehler, 2025-01-29:
@klayoutmatthias Thanks, I can't reproduce that anymore.

But I've noticed a crash in the C++ side of the Python module:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Python [21693]
Path:                  /opt/homebrew/*/Python.framework/Versions/3.12/Resources/Python.app/Contents/MacOS/Python
Identifier:            org.python.python
Version:               3.12.7 (3.12.7)
Code Type:             ARM-64 (Native)
Parent Process:        pycharm [87448]
Responsible:           pycharm [87448]
User ID:               501

Date/Time:             2025-01-29 19:29:41.5030 +0100
OS Version:            macOS 15.2 (24C101)
Report Version:        12
Anonymous UUID:        7ACC797B-51F1-BF61-2F14-6473FFCEB8C2

Sleep/Wake UUID:       F673DB7D-86F2-4F92-BE5F-B3032BE5100B

Time Awake Since Boot: 750000 seconds
Time Since Wake:       8186 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [21693]

VM Region Info: 0 is not in any region.  Bytes before following region: 4375150592
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104c78000-104c7c000    [   16K] r-x/r-x SM=COW  /opt/homebrew/*/Python.framework/Versions/3.12/Resources/Python.app/Contents/MacOS/Python

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   lib_db.cpython-312-darwin.so  	       0x10bf97e38 db::generic_shape_iterator<db::polygon<int>>::operator*() const + 4 (dbGenericShapeIterator.h:368) [inlined]
1   lib_db.cpython-312-darwin.so  	       0x10bf97e38 0x10bf4c000 + 310840
2   lib_db.cpython-312-darwin.so  	       0x10bf97dc8 db::generic_shape_iterator_with_properties_delegate<db::polygon<int>>::generic_shape_iterator_with_properties_delegate(db::generic_shape_iterator<db::polygon<int>> const&) + 60 (dbGenericShapeIterator.h:442) [inlined]
3   lib_db.cpython-312-darwin.so  	       0x10bf97dc8 0x10bf4c000 + 310728
4   lib_db.cpython-312-darwin.so  	       0x10bf88cd8 db::generic_shape_iterator<db::object_with_properties<db::polygon<int>>> db::make_wp_iter<db::polygon<int>>(db::generic_shape_iterator<db::polygon<int>>&&) + 20 (dbGenericShapeIterator.h:505) [inlined]
5   lib_db.cpython-312-darwin.so  	       0x10bf88cd8 void db::region_cop_with_properties_impl<db::edge<int>>(db::AsIfFlatRegion*, db::Shapes*, db::CompoundRegionOperationNode&, db::PropertyConstraint) + 304 (dbAsIfFlatRegion.cc:1023) [inlined]
6   lib_db.cpython-312-darwin.so  	       0x10bf88cd8 db::AsIfFlatRegion::cop_to_edges(db::CompoundRegionOperationNode&, db::PropertyConstraint) + 380 (dbAsIfFlatRegion.cc:1055)
7   lib_db.cpython-312-darwin.so  	       0x10c821e2c db::Region::cop_to_edges(db::CompoundRegionOperationNode&, db::PropertyConstraint) + 80 (dbRegion.cc:269)
8   lib_db.cpython-312-darwin.so  	       0x10d2fc530 gsi::complex_op(db::Region*, db::CompoundRegionOperationNode*, db::PropertyConstraint) + 100 (gsiDeclDbRegion.cc:963)
9   lib_db.cpython-312-darwin.so  	       0x10d329244 0x10bf4c000 + 20828740
10  lib_pya.cpython-312-darwin.so 	       0x1084744b0 0x10845c000 + 99504
11  Python                        	       0x1054e77d8 method_vectorcall_VARARGS_KEYWORDS + 148
12  Python                        	       0x1055d0e94 _PyEval_EvalFrameDefault + 43532
13  Python                        	       0x1055c6218 PyEval_EvalCode + 184
14  Python                        	       0x1055c2a04 builtin_exec + 416
15  Python                        	       0x10552d4fc cfunction_vectorcall_FASTCALL_KEYWORDS + 92
16  Python                        	       0x1055d0e94 _PyEval_EvalFrameDefault + 43532
17  Python                        	       0x1055c6218 PyEval_EvalCode + 184
18  Python                        	       0x105627cb0 run_eval_code_obj + 88
19  Python                        	       0x105625d48 run_mod + 132
20  Python                        	       0x105625258 pyrun_file + 156
21  Python                        	       0x105624638 _PyRun_SimpleFileObject + 288
22  Python                        	       0x105624284 _PyRun_AnyFileObject + 80
23  Python                        	       0x105649808 pymain_run_file_obj + 164
[... 36 lines cut ...]
1   libsystem_pthread.dylib       	       0x19ea2c894 _pthread_cond_wait + 1204
2   Python                        	       0x1056391f4 PyThread_acquire_lock_timed + 284
3   Python                        	       0x1056a3498 acquire_timed + 292
4   Python                        	       0x1056a3064 lock_PyThread_acquire_lock + 56
5   Python                        	       0x1054e77d8 method_vectorcall_VARARGS_KEYWORDS + 148
6   Python                        	       0x1055d0e94 _PyEval_EvalFrameDefault + 43532
7   Python                        	       0x1054de95c method_vectorcall + 372
8   Python                        	       0x1056a28ac thread_run + 144
9   Python                        	       0x105638e2c pythread_wrapper + 48
10  libsystem_pthread.dylib       	       0x19ea2c2e4 _pthread_start + 136
11  libsystem_pthread.dylib       	       0x19ea270fc thread_start + 8

[... cut ...]
   x16: 0x00000000cf72f18f  x17: 0x000000000000018e  x18: 0x0000000000000000  x19: 0x0000600002f85080
   x20: 0x0000600002f85090  x21: 0x000000013bf85810  x22: 0x000060000110d7a0  x23: 0x0000600002f84f00
   x24: 0x0000600002f85080  x25: 0x0000000000000000  x26: 0x0000600003aa2670  x27: 0x7fffffffffffffff
   x28: 0x3ffffffffffffffe   fp: 0x000000016b181ef0   lr: 0x000000010bf97dc8
    sp: 0x000000016b181ea0   pc: 0x000000010bf97e38 cpsr: 0x40001000
   far: 0x0000000000000000  esr: 0x92000006 (Data Abort) byte read Translation fault

```

martinjankoehler, 2025-01-29:
Hi @klayoutmatthias, 
I've made I clean checkout and recompiled everything with the latest `devel` branch.

I still get the same error with the 
```
kpex --pdk sky130A --gds testdata/designs/sky130A/test_patterns/single_plate_100um_x_100um_li1_over_substrate.gds.gz --out_dir=output_sky130A --2.5D
# NOTE: I've removed the need for the y (before it was "--2.5D y")
```

klayoutmatthias, 2025-01-30:
Hmm ... 

I am using kpex at 2adcbdad457e6e9a3ce3e9241ed1a8ce9aafac09 (main head)

I am not using kpex from the wheel (poetry), but I am trying to use if from sources. Which almost works, but needs to manual copy. Here is my recipe:

I build a whl with klayout 0.30 devel branch with ""python3 setup.py bdist_wheel". My system is Ubuntu 24.04.

```
. .venv/bin/activate
pip3 install rich install rich_argparse google protobuf
# this is my build of the "devel branch":
# done with: "python3 setup.py bdist_wheel":
pip3 install klayout-0.30.0-cp312-cp312-linux_x86_64.whl
cd kpex/
./build.sh debug
./build/kpex_Debug/gen_tech_pb 
cp build/sky130A_tech.pb.json klayout_pex_protobuf
cd klayout_pex
ln -s ../pdk
cd ..
```

To run kpex, I have klayout 0.29.11 (Ubuntu 24 public DEB package) in the path.

This is my call:

```
./kpex.sh --pdk sky130A --gds testdata/designs/sky130A/test_patterns/single_plate_100um_x_100um_li1_over_substrate.gds.gz --out_dir=output_sky130A --2.5D
```

and this is my output:

```
────────────────────────────────────────────────────── Command line arguments ───────────────────────────────────────────────────────
/home/matthias/versuche/PEX/kpex/klayout_pex/__main__.py --pdk sky130A --gds                                                         
testdata/designs/sky130A/test_patterns/single_plate_100um_x_100um_li1_over_substrate.gds.gz --out_dir=output_sky130A --2.5D          
─────────────────────────────────────────────────────────── Input Layout ────────────────────────────────────────────────────────────
[20:20:06] INFO     GDS input file passed, running in LVS mode                                                        kpex_cli.py:296
[20:20:06] INFO     No explicit top cell specified, using top cell 'single_plate_100um_x_100um_li1_over_substrate'    kpex_cli.py:332
[20:20:06] INFO     LVS input schematic not specified (argument --schematic), using dummy schematic                   kpex_cli.py:379
─────────────────────────────────────────────────────────── Prepare LVSDB ───────────────────────────────────────────────────────────
[20:20:06] INFO     Cache miss: extracted LVSDB does not exist                                                        kpex_cli.py:712
output_sky130A/.kpex_cache/sky130A/home/matthias/versuche/PEX/kpex/testdata/designs/sky130A/test_patterns/single_plate_100um_x_100um_
li1_over_substrate.gds.gz/single_plate_100um_x_100um_li1_over_substrate.lvsdb.gz                                                     
[20:20:06] INFO     Calling klayout -b -r                                                                            lvs_runner.py:70
                    /home/matthias/versuche/PEX/kpex/klayout_pex/pdk/sky130A/libs.tech/kpex/sky130.lvs -rd                           
                    input=/home/matthias/versuche/PEX/kpex/testdata/designs/sky130A/test_patterns/single_plate_100um                 
                    _x_100um_li1_over_substrate.gds.gz -rd                                                                           
                    report=/home/matthias/versuche/PEX/kpex/output_sky130A/single_plate_100um_x_100um_li1_over_subst                 
                    rate__single_plate_100um_x_100um_li1_over_substrate/single_plate_100um_x_100um_li1_over_substrat                 
                    e.lvsdb.gz -rd                                                                                                   
                    schematic=/home/matthias/versuche/PEX/kpex/output_sky130A/single_plate_100um_x_100um_li1_over_su                 
                    bstrate__single_plate_100um_x_100um_li1_over_substrate/single_plate_100um_x_100um_li1_over_subst                 
                    rate_dummy_schematic.spice -rd thr=22 -rd run_mode=deep -rd spice_net_names=true -rd                             
                    spice_comments=false -rd scale=false -rd verbose=true -rd schematic_simplify=false -rd                           
                    net_only=false -rd top_lvl_pins=true -rd combine=false -rd combine_devices=false -rd purge=false                 
                    -rd purge_nets=false -rd no_simplify=true, output file:                                                          
                    output_sky130A/single_plate_100um_x_100um_li1_over_substrate__single_plate_100um_x_100um_li1_ove                 
                    r_substrate/single_plate_100um_x_100um_li1_over_substrate_lvs.log                                                
─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
2025-01-30 20:20:09 +0100: Memory Usage (635664K) : Starting running SKY130 Klayout LVS runset on                                    
/home/matthias/versuche/PEX/kpex/testdata/designs/sky130A/test_patterns/single_plate_100um_x_100um_li1_over_substrate.gds.gz         
2025-01-30 20:20:09 +0100: Memory Usage (635664K) : Layout file:                                                                     
/home/matthias/versuche/PEX/kpex/testdata/designs/sky130A/test_patterns/single_plate_100um_x_100um_li1_over_substrate.gds.gz         
2025-01-30 20:20:09 +0100: Memory Usage (635664K) : Number of threads: 22                                                            
2025-01-30 20:20:09 +0100: Memory Usage (635664K) : Final report:                                                                    
[... cut ...]
C1;PLATE;VSUBS;386.18                                                                                                                
─────────────────────────────────────────────────────── Extracted netlist CSV ───────────────────────────────────────────────────────
/home/matthias/versuche/PEX/kpex/output_sky130A/single_plate_100um_x_100um_li1_over_substrate__single_plate_100um_x_100um_li1_over_su
bstrate/single_plate_100um_x_100um_li1_over_substrate_k25d_pex_netlist.csv                                                           
```

No segfault :(

Matthias

martinjankoehler, 2025-01-30:
It might have had something to do with my Python versions.
Somehow, an update script did change the `PATH` in my `.zprofile` was changed to force the system Python 11 version.
With my venv and python 3.12 in the `PATH` corrected, it now works.

closed 2025-02-13 by martinjankoehler