KLayout/klayout #1983: Can the file parsing module be encapsulated into an independent library?

libaineu2004, 2025-02-06:
KLAYOUT is a great project. The file parsing module is very well done.
I hope that the file parsing module and file format conversion module in the project can be separated. Independently become a new C++ project. A lightweight file parsing and conversion library.
For example:
This library can open dxf files and save them as GDS2 Text files.
Define the relevant interfaces,
class CFileModule
{
public:
bool opendxf(QString dxfpath);
bool convert(QString dxfpath, QString gds2txtpath);
};

klayoutmatthias, 2025-02-09:
No, a C++ API is not on the roadmap. 

But for a lightweight solution, you can take a look at the Python klayout module: https://pypi.org/project/klayout/

Matthias

libaineu2004, 2025-02-11:
> No, a C++ API is not on the roadmap.
> 
> But for a lightweight solution, you can take a look at the Python klayout module: https://pypi.org/project/klayout/
> 
> Matthias

Can Python API realize:
1. Read dxf file
2. Convert gds2 text file
Can you write an example?
Thank you

chkkbim, 2025-02-11:
It is best if you can ask this question in the forum [https://www.klayout.de/forum/].
There are more people who can help you there.