KLayout/klayout #1984: Latest version of klayout crashes at startup when C4M/PDK/* salt packages are installed

stafverhaegen-chipflow, 2025-02-13:
I'm using klayout 0.29.11 on Linux Mint 22. I'm using the Ubuntu 24.04 deb package for installation as the Linux Mint version is derived from that distro.
When I install the C4M/PDK salt packages klayout crashes on startup the next time it is started. Attached is the crash log.
I will need some guidance to know if the error is in my salt packages or is on klayout side.

[klayout_crash.log](https://github.com/user-attachments/files/18784820/klayout_crash.log)

stafverhaegen-chipflow, 2025-02-13:
The above was still with a previous .klayout. I now started with a full new .klayout (e.g. removed) and get the same crash log after installing my salts.

stafverhaegen-chipflow, 2025-02-13:
Tried with version 0.29.7 and got same behaviour.

stafverhaegen-chipflow, 2025-02-13:
I'm using latest release of libc6: 2.38-0ubuntu8.4
Also tried klayout version 0.29.9-1+9.1 as packaged by Ubuntu and this has same problem.

klayoutmatthias, 2025-02-16:
I can reproduce the issue, but debugging just takes me here:

[image]

There are no KLayout calls involved so far. I have no clue what goes on. The stack trace is from the Python interpreter only, except the initial call into it.

klayoutmatthias, 2025-02-16:
Ah ... looks like we got a Python issue here.

I packed all the relevant files into this: 

[issue-1984.tar.gz](https://github.com/user-attachments/files/18816649/issue-1984.tar.gz)

With these files

```
$ cd issue-1984
$ python3 register_prim_lib.py
```

Which gives a segfault on my Python 3.12.3 on Ubuntu 24.04.

There is no KLayout involved - I can run this from a fresh venv and it still crashes.

Matthias

klayoutmatthias, 2025-02-16:
The bug is already reported: https://github.com/python/cpython/issues/118814

So not my fault :)

klayoutmatthias, 2025-02-16:
The fix is to specify the name as positional parameter:

```
_cell_type_ = TypeVar("_cell_type_", bound="Cell")
_factory_type_ = TypeVar("_factory_type_", bound="CellFactory")
_factorycell_type_ = TypeVar("_factorycell_type_", bound="FactoryCell")
_factorycell_type2_ = TypeVar("_factorycell_type2_", bound="FactoryCell")
```

klayoutmatthias, 2025-02-17:
I'll close this issue, if you don't mind, as it's a Python problem.

Thanks,

Matthias