KLayout/klayout #1985: KLayout Python Module - Wiki - not working links

pmierzwi, 2025-02-13:
The links are not working.

klayout.tl module: [http://www.klayout.org/downloads/master/doc-qt4/code/module_tl.html](http://www.klayout.org/downloads/pymod/doc-qt4/code/module_tl.html)
    klayout.db module: [http://www.klayout.org/downloads/master/doc-qt4/code/module_db.html](http://www.klayout.org/downloads/pymod/doc-qt4/code/module_db.html)
    klayout.rdb module: [http://www.klayout.org/downloads/master/doc-qt4/code/module_rdb.html](http://www.klayout.org/downloads/pymod/doc-qt4/code/module_rdb.html)

The correct (but not-sure) links:

https://www.klayout.de/doc-qt5/code/module_db.html
https://www.klayout.de/doc-qt5/code/module_rdb.html
https://www.klayout.de/doc-qt5/code/module_tl.html

Regards

klayoutmatthias, 2025-02-13:
Thanks for reporting this issue. 

Qt4 is deprecated and the "master" links are CD-generated. The last links are the valid ones.

However, which Wiki do you refer to?

Matthias

klayoutmatthias, 2025-02-13:
Ah, sorry. Got it. I forgot about the Wiki.

Matthias

klayoutmatthias, 2025-02-13:
Maybe I should delete this page. It was meant to inform users during the creation of the Python modules and it is mostly outdated. Now everything is on PyPI: https://pypi.org/project/klayout/ and there is some entry-level documentation with working links here: https://www.klayout.org/klayout-pypi/

Matthias