KLayout/klayout #1997: Can not find a file in Open Recent menu.

masukatto0281, 2025-03-05:
If the path contains certain Japanese characters, they will not be recognized correctly in the “Open Recent” menu.
The file can be opened, but is not recognized as existing.

To Reproduce:
1. Make folder naming `あ1い2う3え4お5`.
2. Place *.gds file.
3. Open gds fle in KLayout.
4. Look Open Recent menu.

I feel this is a character code conversion issue.

Screenshot:
[image]
`C:\Users\<UserName>\あ1い2う3え4お5\<FileName>.gds`

Versions:
KLayout 0.29.11 (2025-01-18 r18c86cb20)
Windows 11 24H2 (26100.3194)

Thank you.

Kazzz-S, 2025-03-05:
Hello @klayoutmatthias,

I could reproduce this issue (related to #1213) on Mac, as shown below.
I've attached the test data. -->  [gh1997-test.zip](https://github.com/user-attachments/files/19081532/gh1997-test.zip)

[image]

[image]

It looks like the problem appears when a directory name has mixed ASCII and non-ASCII characters.

Best regards,
Kazzz-S

klayoutmatthias, 2025-03-05:
I reproduce and fix this issue :)

Thanks for reporting. I wonder why this was not seen before.

The fix will go into 0.29.12.

Best regards,

Matthias

Kazzz-S, 2025-03-05:
Hi @klayoutmatthias,

Thanks for a quick fix. I confirmed it :-)

[image]

Best regards,
Kazzz-S

klayoutmatthias, 2025-03-06:
Thanks for confirming this :) 

I have released 0.29.12 with the fix included.

Matthias

closed 2025-03-06 by klayoutmatthias

masukatto0281, 2025-03-21:
Ver. 0.29.12 confirmed that the problem has been resolved.

Thank you very much for your help.