KLayout/klayout #1998: 0.29.12 python package only released as tar.gz

sebastian-goeldi, 2025-03-07:
Hi Matthias,

I think 0.29.12 got pushed only as a `tar.gz` and not any wheels. Did maybe pypi run out of space for the package again?

Best,
Sebastian

joamatab, 2025-03-07:
See https://pypi.org/project/klayout/#files

I think the problem is with download artifact
https://github.com/KLayout/klayout/actions/runs/13705050468/job/38333103526

klayoutmatthias, 2025-03-13:
I had to switch to upload/download-artifact v4 as they have deprecated v3 finally.

Maybe something is wrong with that version.

I'll try rerunning, but GitHub actions are so difficult to debug.

Matthias

sebastian-goeldi, 2025-03-13:
Hi Matthias,

I see. Please don't waste too much time on it. I can dig into it tomorrow. I am pretty sure the issue is some artifact name or similar.

Best,
Sebastian

sebastian-goeldi, 2025-03-13:
Also, maybe for now the easiest path is to manually upload the artifacts. I believe you can just grap them from the pipeline with the upload artifact (scroll down to the bottom and there should be the files)

sebastian-goeldi, 2025-03-13:
Ah yes,

It is very likely a problem of v4 not allowing the job names to be the same (facepalming really hard here about that choice...)

https://github.com/actions/upload-artifact/blob/main/docs/MIGRATION.md

sebastian-goeldi, 2025-03-13:
Or it needs the merge step, not so 100% sure

klayoutmatthias, 2025-03-13:
From what I see it was just pulling the sdist part. I remove the names from the download section and gave the sdist artifact a different name. Maybe that fixes it.

Windows builds stopped working on Azure after they switched to setuptools 76.0. I need to debug the issue. Apparently my setup.py no longer is compatible with setuptools. That was to be expected, but this deployment stuff is really eating too much of my time.

Matthias

klayoutmatthias, 2025-03-14:
I love debugging foreign tools: https://github.com/pypa/setuptools/issues/4885

I will pin the version of setuptools now to <76.0.0 and try again.

klayoutmatthias, 2025-03-14:
I have some hope I fixed things now. Let's see tomorrow morning.

klayoutmatthias, 2025-03-14:
Finally we're there :)

Hope there are no new issues on the next release!

Matthias

closed 2025-03-14 by klayoutmatthias