KLayout/klayout #2004: 0.30 - `rdb.RdbItem.add_value` should accept the new `PolygonWithProperties` (and other `[Shape]WithProperties`)

sebastian-goeldi, 2025-03-25:
Hi Matthias,

First of all, thanks for the great additions with klayout 0.30 !

The current RdbItem does not accept the new types, when I feel it should (or at least downcast it internally first). Should I find more problems, I will update below.

Best,
Sebastian

sebastian-goeldi, 2025-03-25:
Ok, found another one (not necessarily a bug but inconsistency):

`Region[index]` returns a `Polygon` vs `Region.each()` returns `PolgyonWithProperties` (the latter is what the initial error triggered, the former I found while experimenting)

klayoutmatthias, 2025-03-25:
Hi Sebastian,

thanks for this report. Please go ahead and look for more problems :)

Usually downcasting is done automatically, but with "add_value" there are two paths that are accessible with downcast - one through `add_item(DPolygon)` and another with `add_item(RdbItemValue(DPolygon))`. That leads to the ambiguity. It's easy to fix as basically only the second path is needed for all types.

I don't know how to represent properties in RDB values, so I guess best is to skip them.

Matthias

klayoutmatthias, 2025-03-25:
Here is a problem I found:

* DEdgePairWithProperties is derived from EdgePair instead of DEdgePair

Matthias

klayoutmatthias, 2025-03-26:
Plus another issue: 

* `Texts[n]`, `Edges[n]` and `EdgePairs[n]` does not deliver objects with properties at all.

closed 2025-04-05 by klayoutmatthias