KLayout/klayout #2007: Layer map hits parse error with layer that starts with a digit

jeffng-or, 2025-03-31:
LayerMap object errors out when encountering a layer mapping entry that has a layer name starting with a digit:

[image]

Here's the code to repro:
```
#!/usr/bin/env -S klayout -r

layer_name = "1Layer"
layout = RBA::Layout::new
cell = layout.create_cell("TOP")
layer = layout.layer(layer_name)

if layer
  puts "Able to create layer " + layer_name
end

map_str = layer_name + ":123/456"
layer_map = RBA::LayerMap::new
layer_map.map(map_str)
```

stefanottili, 2025-04-01:
layer names starting with digits are highly unusual, what technology is this ?

jeffng-or, 2025-04-01:
It's a foundry-provided, proprietary PDK.

jeffng-or, 2025-04-02:
got it. Wrapping the layer name with a single quote appears to work

closed 2025-04-02 by jeffng-or