KLayout/klayout #2009: Enable layer mapping of LefLabel object

jeffng-or, 2025-03-31:
When LEF import encounters a LEF PIN, it also creates a LefLabel object containing the pin's name, which is nice. If no layer mapping is applied, the label is accessible from the GUI.

However, if I want to map layers, there's no support to be able to map the LEFLABEL to a gds layer/datatype, so you lose the visibility/select ability of the label object. Here's the warning message:

```
Warning: Reading layer map file /home/user/label.map, line 3: purpose LEFLABEL ignored for layer metal1
```

In other words, I want to be able to use a layer mapping file like:

```
metal1 NET 17 0
metal1 LEFPIN 17 24
metal1 LEFLABEL 17 5
```
 and be able to view/select the net, pin, and label shapes in the GUI.

stefanottili, 2025-03-31:
strm2oas -h  shows the syntax for the --lefdef-map.
For labels use "NAME metal1/PIN 17 5"

Note that some technologies, e.g. sky130 require two LEFPIN statements.
One of the net geometry of the pin and one for the pin geometry.
One pin, two rectangles on different layer:datatype.

klayoutmatthias, 2025-04-01:
Got the message about the missing LEFPIN documentation :)

jeffng-or, 2025-04-02:
Thanks for the info! I was able to get access to the pin text as LEFLABELS by using the NAME syntax.

closed 2025-04-02 by jeffng-or

Super-Karl, 2025-04-29:
> Thanks for the info! I was able to get access to the pin text as LEFLABELS by using the NAME syntax.

@jeffng-or Hi! Have you solve the problem? I meet the same problem. I tired both "Metal1 LEFLABEL 17 5" and "NAME Metal1/LEFLABEL 17 5". But it didn't work in my testcase, and I got the same warning. What your version? Can you provide the whole map file?

stefanottili, 2025-04-29:
@Super-Karl See my previous reply.
For lef labels use "NAME metal1/LEFPIN 17 5".