KLayout/klayout #2012: Shapes object cannot be garbage collected

klayoutmatthias, 2025-04-01:
This ticket refers to this discussion:

https://www.klayout.de/forum/discussion/2687/shapes-object-cannot-be-garbage-collected#latest

klayoutmatthias, 2025-04-03:
@sebastian-goeldi I guess there is a test suite for GDSFactory or KFactory I could run to verify that I have not broken anything. This is quite a basic code change. Do you have a pointer for me? 

Thanks.

Matthias

sebastian-goeldi, 2025-04-03:
Hi Matthias,

Yes absolutely. Clone gdsfactory, `python -m pip install -e .[dev]` and then `pytest -n logical` (`-n logical` here will use all cores available, but you can also run them single threaded by just leaving that away, the outputs for errors are slightly better then)

klayoutmatthias, 2025-04-04:
Perfect. Thanks!

Works like a charm and I only get the same warnings than with the released version (related to bend_euler_all_angle mostly) :)

So I'm confident I did not kill anything.

Thanks again and best regards,

Matthias

sebastian-goeldi, 2025-04-04:
Great, thanks a lot! Yeah, when I saw the issue, I thought this will be very important. Especially when using the file watcher. That one regularly cleans cell caches, so potentially this could cause the memory to balloon fast. I am sure this will help a lot.

closed 2025-04-05 by klayoutmatthias