KLayout/klayout #2014: strm2oas lef/def/gds collect drops cells referenced by sky130 spare

stefanottili, 2025-04-02:
Test-case sent directly to Matthias.

stefanottili, 2025-04-02:
It seems my mail got stuck in my outbox, so here's the tarfile of the test-case.
[test_strm2oas.tgz](https://github.com/user-attachments/files/19576542/test_strm2oas.tgz)

klayoutmatthias, 2025-04-05:
I had not considered that case that a COMPONENT replacement cell would use other COMPONENT replacements as child cells. This is a more basic issue of the "create_multi_mapping" function.

klayoutmatthias, 2025-04-05:
I see your other point regarding the XOR results: one shape appears in the XOR output on layer 236/0 which is a false positive. That is a bug in the optimization we introduced: A cell is skipped when entirely covered by a box. That needs variant building as in one cell this is the case while in the other it isn't.

I will try to include a fix in this ticket too.

Matthias