KLayout/klayout #2016: Lazy evalution of PCell created from shape

romainSW, 2025-04-04:
Dear,
 
We are using design where some PCell are created from a Path (and depend on its geometry from the path). As they are pretty computation intensive to generate, the PCell are lazy evaluated when we modified their parameters. 

However, when we modify the underlining Path, the evaluation of the PCell is not lazy (it immediately try to update the PCell at every modification of the Path). Will it be possible to make the generation of the PCell lazy when editing shape from which it is depending, please ?

Thank you again for this amazing software.

Romain A

klayoutmatthias, 2025-04-04:
Hi Romain,

yes, I think that's possible. I simply did not consider that use case, but it makes sense.

Thanks for bringing this up! 

Matthias

klayoutmatthias, 2025-04-05:
I just checked the implementation, and it's already lazy in a way, but not in all.

So while you are dragging a vertex of a guiding shape in partial mode for example, the update only happens once you placed the vertex. But not during dragging.

I think you refer to the path properties widget. When you update the path properties there - for example, by editing the width - the path (and PCell) is updated immediately. 

It that what you refer to?

BR,

Matthias

klayoutmatthias, 2025-04-05:
I have implemented a first patch. It will disable immediate update of the PCell during edits of the shape properties in the property pages. 

Maybe you have a way to check this implementation.

Matthias

romainSW, 2025-04-07:
Hi Matthias,

You were right, I was referring to the path properties widget (manually editing the coordinate of the path in this widget). 

I compiled this morning the code on a debian bookworm (all library per default) in the relevant branch and test it. It works without issue during my short testing !

Sadly, I will not be able to do extensive testing as it is not my main computer and doing design remotely is not very practical in that case. I do not think that they are nightly compilation for windows but they are, I can do some testing on windows 11 too.

Thanks a lot for developing this enhancement !

Best regards.

Romain

klayoutmatthias, 2025-04-29:
The new version 0.30.1 is released, so I am closing this ticket.

Thanks!

Matthias

klayoutmatthias, 2025-05-04:
I stand corrected. Unfortunately the solution was not released with 0.30.1. Instead it will go into 0.30.2.

Sorry for this.

Matthias

closed 2025-05-22 by klayoutmatthias