KLayout/klayout #2019: build failure agaisnt Qt6.9: error: ‘class QDomNodeList’ has no member named ‘operator!=’ and 'operator=='

Kazzz-S, 2025-04-06:
This error may be related to #1987.

* macOS Sonoma (14.7.5)
* klayout 0.30.0
* commit 449a9a96
* Qt 6.9.0 Homebrew
* Apple clang version 15.0.0 (clang-1500.3.9.4)

```
:
../../../../src/gsiqt/qt6/QtXml/gsiDeclQDomNodeList.cc:171:50: error: no member named 'operator!=' in 'QDomNodeList'
  ret.write<bool > ((bool)((QDomNodeList *)cls)->operator!= (arg1));
                          ~~~~~~~~~~~~~~~~~~~~~  ^
../../../../src/gsiqt/qt6/QtXml/gsiDeclQDomNodeList.cc:209:50: error: no member named 'operator==' in 'QDomNodeList'
  ret.write<bool > ((bool)((QDomNodeList *)cls)->operator== (arg1));
                          ~~~~~~~~~~~~~~~~~~~~~  ^
2 errors generated.
:
```

[qt6Brew.build.macos-Sonoma-release-Rhb34Phb312.log.gz](https://github.com/user-attachments/files/19621053/qt6Brew.build.macos-Sonoma-release-Rhb34Phb312.log.gz)

Kazzz-S

klayoutmatthias, 2025-04-09:
Thanks for pointing this out.

I had to build Qt 6.9 from sources first to reproduce the problem. It looks like it's a single place only in QDomNodeList. That's hopefully easy to fix.

I'll keep you updated.

Best regards,

Matthias

Kazzz-S, 2025-04-09:
Thank you for the prompt resolution!
I successfully built the 0932944 branch.

[image]
[image]

Best regards,
Kazzz-S

closed 2025-04-27 by klayoutmatthias