KLayout/klayout #2021: Internal error in dbManager

romainSW, 2025-04-09:
Hi,

During the generation of multiple PCell, I have from time to time the following error which happens : 
`ERROR: Internal error: ../../../src/db/db/dbManager.cc:376 m_opened was not true` 

To explain more precisely, I am using a PCell which generated a lot (in the few thousands) of PCell (of all of the same variant). When changing some of the parameters of the generating PCell, sometimes this errors occurs but it is not very reproducible.

This bug occurs at least with klayout 30.0 and 29.12 under windows 11.

I can privately provide the file (design and PCell) as well as a log generated with debug option 41. If you are interested, where should I send them? Will it be useful to get a windows memory dump of the program too?

I tried to reproduce the bug with branch feature/issue-2016 under debian bookworm but I could not get an error after 30mins of testing (when I typically trigger the bug in window every few mins). If you think that it is useful to test it with another version of klayout under linux, I can do it too.

I am available to perform any testing that you think will help with the situation.

Best

Romain

klayoutmatthias, 2025-04-09:
Please check if the problem also occurs when the macro IDE is not open or if you can enabled debugging in the IDE.

There is a stability issue when you develop PCells with the Macro IDE as the code execution triggers callbacks which itself trigger events inside the parameter dialogs. In some cases that can lead to interferences that trigger this bug.

The only way to prevent this problem consistently is to disable the IDE or to delegate that to a separate process with it's own event loop. I am trying to fix these kind of issues when I come across them myself, but there are too many rogue execution paths. 

After all, my reasoning is that PCells should be stable when you use them - i.e. without the IDE open. If you observe these problems with the IDE closed, that's a different issue. If only with the IDE open, I'd say that is a kind of known issue, but there is no magic fix.

Matthias

klayoutmatthias, 2025-04-12:
No worries, thanks for bringing up this issue.

It reminds me that I need to keep on watching this problem and eliminate execution paths that cause this issue.

BR,

Matthias