KLayout/klayout #2024: Feature Request: Make grid density threshold configurable

JarmoHu, 2025-04-15:
**Current Behavior:**
When zooming in, the displayed major grid spacing is dynamically adjusted based on the grid's pixel size (`dgrid`) relative to the fixed font width (`fwr`). According to the code in `layGridNet.cpp` (`render_bg` function), the threshold for using a coarser grid (and potentially switching to `style2`) seems to be hardcoded around `dgrid < fwr * 4`. This means the grid becomes relatively sparse even when there might be sufficient screen space to display a denser grid.

**Desired Behavior:**
I would like the multiplier used in this threshold (currently `4`) to be configurable by the user. This would allow users to choose a smaller threshold (e.g., `1`, `2`) if they prefer to see a denser grid at higher zoom levels.

**Why this feature is needed:**
For tasks involving very fine layout details or on high-resolution displays, the current hardcoded threshold can limit the usefulness of the grid as a visual reference at high magnification. Being able to adjust this threshold would provide more flexibility to match user preferences and specific workflow needs. For example, I often find the grid disappears or becomes too coarse earlier than I'd like when inspecting nanoscale features.

**Context:**
*   KLayout Version: 0.30.0
*   Operating System: win10 20h2
* 
**Suggested Solution:**
Add a configuration option, possibly in the "Setup -> Display -> Background" settings page, allowing users to set the "Grid density font width multiplier" (or a similar name), with a default value of 4.

Thank you for considering this enhancement request!

klayoutmatthias, 2025-04-24:
Hi @JarmoHu,

thanks for this suggestion.

In my case, the 4x factor already creates pretty dense grids. This is how dense the grid becomes and how that relates to the font size:

[image]

Maybe you can share a screenshot how it looks in your case?

Matthias

klayoutmatthias, 2025-04-26:
That's how it looks now

[image]

JarmoHu, 2025-04-26:
> That's how it looks now
> 
> [image]

Thanks very much.