KLayout/klayout #2025: 0.30: `TypeList` PCell parameter issue in Instance Properties dialog

caspar-iqm, 2025-04-22:
Starting with version 0.30.0 (testing on Windows), list parameters get surrounded in parentheses in the Instance Properties dialog. Every time any parameter is changed, another set of parentheses is added around the list.

From what I can see, this issue is related to the properties window. I did not notice issues using PCells programatically.

### Example

A basic PCell with list parameter:

```py
import pya

class DummyPCell(pya.PCellDeclarationHelper):
  def __init__(self):
    super(DummyPCell, self).__init__()

    self.param("s", self.TypeShape, "", default = pya.DPoint(0, 0))  # Something to select
    self.param("n", self.TypeInt, "Number", default = 10)
    self.param("list_of_strings", self.TypeList, "List of strings", default = ["one", "two", "three"])

  def coerce_parameters_impl(self):
    print(f"{self.n=}, {self.list_of_strings=}")

class MyLib(pya.Library):
  def __init__(self):
    self.description = "Dummy Library"
    self.layout().register_pcell("DummyPCell", DummyPCell())
    self.register("MyLib")

MyLib()
```

Placing an instance and opening the properties dialog:

[image]

Editing the `n` property or even just going through the list with tab adds more parentheses:

[image]

The printout in `coerce_parameters_impl` shows that when reading the dialog, the parentheses just get interpreted as part of the first and last item:

```
self.n=1, self.list_of_strings=['one', 'two', 'three']
self.n=1, self.list_of_strings=['(one', 'two', 'three)']
self.n=1, self.list_of_strings=['((one', 'two', 'three))']
self.n=1, self.list_of_strings=['(((one', 'two', 'three)))']
```

caspar-iqm, 2025-04-22:
A stop-gap workaround for the example above is to just keep remove the parenthesis every time. Of course this breaks if there are actual intended parenthesis inside the list items. 

```py
  def coerce_parameters_impl(self):
    list_of_strings = self.list_of_strings
    if len(list_of_strings) > 0:
       list_of_strings[0] = list_of_strings[0].removeprefix("(")
       list_of_strings[-1] = list_of_strings[-1].removesuffix(")")
       self.list_of_strings = list_of_strings
```

With this, a single level of parentheses keeps visible in the GUI (so they are probably added when filling the text field)

klayoutmatthias, 2025-04-24:
Oh, that's weird.

Thanks for reporting the problem. I will fix it.

Matthias

klayoutmatthias, 2025-04-24:
That issue reveals a somewhat deeper problem: while the display wants to preserve the array of strings as much as possible, the reverse translation does a dumb split at comma characters.

Specifically that means: 
* you can't enter strings will commas
* you can't enter anything else than strings
* you cannot enter an empty list

So my favorite solution is (see branch):
* if the list elements are plain strings without a comma, revert to the original behavior
* an empty string is not a list with one empty string member, but an empty list now
* complex lists (such with numbers, nested ones or containing strings with comma), use a more elaborate notation using the round bracket as an indicator for that format

Samples for the more elaborate format are:

````
# A list of two strings, one containing a comma:
('a,b', 'c')
# two integer numbers:
(#1,#2)
# nested lists:
('a',('b','c'))
```

If you don't need all of that, it is better to use a plain string parameter and do the splitting yourself. However, it's kind of difficult to switch the parameter type when a PCell is already out and used, so I don't think, that is a a good thing to do. But consider the "empty list" case - this simply was not possible before and I think it makes sense to have it. Right?

Matthias

closed 2025-04-27 by klayoutmatthias

caspar-iqm, 2025-04-28:
Very nice to have lists that preserve types, thanks!