KLayout/klayout #2026: after 2.5D display main 2D layout does not display anymore with visible shapes

woelt, 2025-04-24:
steps:
I create a layout with a path shape at level (1350/0)
I crate a 2.5D script:  z(input(10000,0),zstart:0.00,zstop:0.29,name:"prb") 

after I run the 2.5D script and the layer is displayed in the 2.5D view
I cant see the the original 2D shape in the layout window anymore (also I notice that the TOP cell somehow seems to have disappeared from the cell list.

Pieter

klayoutmatthias, 2025-04-24:
Hmm ...

I can't reproduce the issue. Even if I use the right input layer:

```
 z(input(1350,0),zstart:0.00,zstop:0.29,name:"prb")
```

[image]

Also, the normal view and the 2.5d view are not linked. So maybe that is a graphics driver issue? 2.5D view uses OpenGL which often is a cause of trouble.

Matthias

woelt, 2025-04-25:
Interesting so you are implying it is OpenGL? 
I went back to 29.7 and that seems to work. So have a look at all the changes you made in between 30.0 and 29.7....
I'll try it on my personal computer as well and let you know how it goes...

klayoutmatthias, 2025-04-26:
Hold on ... I think I managed to reproduce the problem. I will try to debug it.

Matthias