KLayout/klayout #2027: DRC hangs for width> check

pdumon, 2025-04-25:
On a finely discretized polygon, `layer.drc(width<value)` works fine, but `layer.drc(width>value)` is extremely slow and hangs 'indefinitely'.

[image]

DRC script:

```
report("DRC")
layer1 = input(1, 0)

$stdout.puts "# check <100um"
rule1 = layer1.drc(width<100.0.um)
rule1.output("rule1")

$stdout.puts "# check >100um"
rule2 = layer1.drc(width>100.0.um)
rule2.output("rule2")
```

pdumon, 2025-04-25:
How can I attach the GDSII file used for testing here?

pdumon, 2025-04-25:
I have a workaround, by first snapping the polygon to a coarser grid (reducing its # vertices) I can get it to run - still slow but at least it runs. But this workaround reduced DRC accuracy, so better if the issue could be solved.

klayoutmatthias, 2025-04-26:
You can attach GDS by wrapping it in a .zip archive.

But there is no fix - you're simply putting a too complex problem into the algorithm.

A range of 100µm means the algorithm will collect everything in the vicinity of 100µm of an edge and performs a width check on every pair in that range. That happens for a lot of pairs, so performance goes down. 

Think about using less points to resolve your corner - your resolution is much too high and does not give you any benefit. It's easy to do some estimation of the number of points you need based on the accuracy requirements of your process. Typically, a hundred points are more than enough to represent a circle. This is based on the assumption that a typical manufacturing process has a relative accuracy requirement - that is dimension variation divided by the dimension - of maybe 0.1%, but not much less.

Matthias

pdumon, 2025-04-27:
Agreed. That's what I did by snapping..
No, this is actually for a maximum width rule, which is often stated on eg metals

But interestingly, the same/similar performance problem appears with a much less dense multipolygon: a metal with polygon (but not at all high resolution) perforation just hangs on the same check

pdumon, 2025-04-27:
Thanks, by the way ☺️

klayoutmatthias, 2025-04-27:
The "width greater something" check suffers from a conceptual problem: let's assume you have a maximum width of 10um and you draw a.wire with an width of 1um and length of 20um, then formally there is an axis (along the length) where the width is wider than 10um. But this is not a violation.

On closer look "width greater something" is not a distance check, but requires a more elaborate analysis like "report edges without an opposite one within a distance of 10um, unless edge pairs perpendicular to that are closer than 10um". This is a complex analysis in the generic case. 

Maybe performance can be improved, but there are limits and when it comes to a simple check, there is a more efficient alternative: use a.negative sizing.with half the maximum width. When there is something left, the original polygon was larger than the maximum width.

If you apply a large sizing value (-50 um for a max width of 100um in your case) to complex polygons, the performance will be better if you do that in steps,  e.g. 10 times -5 um. If the steps are chosen in a way that most polygons vanish in the first step, the following steps will only take little time.

Matthias

pdumon, 2025-04-27:
> The "width greater something" check suffers from a conceptual problem: let's assume you have a maximum width of 10um and you draw a.wire with an width of 1um and length of 20um, then formally there [...]

Thanks, Matthias! 
Indeed using sizing for various operations