KLayout/klayout #2032: About the Quickly merge all layers

Eify666666, 2025-04-27:
Hello developer, 
Are you considering designing a function for the software that can merge all the selected layers separately with one click?
Now we have to select layer by layer and then click the merge function

klayoutmatthias, 2025-04-28:
Hi @Eify666666,

thanks, that is actually a useful suggestion.

By "Merge" you mean the "Edit/Layer/Merge" feature, right?

You can basically do a merge also with a DRC script:

```
# This example merges 1/0, 5/0, 7/0 and 10/0 into 99/0:
p = input(1, 0) + input(5, 0) + input(7, 0) + input(10, 0)
p.merged.output(99, 0)
```

Maybe that is more convenient as of now.

Matthias

Eify666666, 2025-04-29:
Hello Matthias,

When I draw each layer, I use a large number of geometric shapes and then stitch them together. When merging,  for example, all the graphics in 1/0 need to be merged tp 1/0, and the same applies to layers such as 5/0, etc. Therefore, I would like to ask if there is a way to merge all the geometric shapes in each layer separately instead of merging all the layers into the same layer. I might have more than 30 layers that need to be merged.

The code you wrote seems to have merged all the layers together.

Thanks,
Eify

klayoutmatthias, 2025-04-29:
Ah .. I see. So it is multiple inputs and multiple outputs what you are looking for.

In DRC that is a loop:

```
[ [1, 0], [5, 0], [7, 0], [10, 0] ].each do |l|
  input(*l).merged.output(*l)
end
```

Matthias

Eify666666, 2025-05-02:
Thanks very much！