KLayout/klayout #2033: klayout-0.30.0-win64 read lef def pin missing

Super-Karl, 2025-04-28:
I use klayout to open lef/def file which is in ispd 2018 contest dataset. The pins are missing. I use the default setting and did not change anything.
Here is the lef def files and the picture.

[ispd18_sample.solution.good.zip](https://github.com/user-attachments/files/19936714/ispd18_sample.solution.good.zip)

[image]

Super-Karl, 2025-04-28:
And I also use 0.28.17 to open the same case. It shows the error.

[image]

Super-Karl, 2025-04-28:
I open it by click "File > Import > DEF/LEF", and here is the final ok picture.

[image]

Super-Karl, 2025-04-29:
Thanks for your reply,
I use klayout as a lef/def viewer. But I can't see the pins inside std cell by klayout-0.30.0-win64. I can see the pin inside std cell by klayout-0.26.12-win64. The 0.30.0 doesn't seem to read all layers and I actually clicked the Show All Layers option.

Both clicked.
[image]

[image]

stefanottili, 2025-04-29:
Use lef/def Options "Always produce LEF geometry and ignore LEF FOREIGN" to get klayout to read the lef MACRO and it's pin/blockages as layout cells.

[image]

stefanottili, 2025-05-01:
@Super-Karl Can you please let us know whether this is working for you ?

Super-Karl, 2025-05-01:
> @Super-Karl Can you please let us know whether this is working for you ?

@stefanottili Yes! I tried it just now. It is working! Thank you very much!!!

stefanottili, 2025-05-01:
Great. Please close the issue, only you and the maintainer can.

closed 2025-05-01 by Super-Karl