KLayout/klayout #2041: strm2oas should read --lefdef-lefs and --lefdef-lef-layouts only once

stefanottili, 2025-05-06:
strm2oas/gds reads all --lefdef-lefs and --lefdef-lef-layouts for each of the def files.

This is unnecessary, they should only be read once.

klayoutmatthias, 2025-05-08:
Understood and I agree. It's just not straightforward to implement :(

Matthias

klayoutmatthias, 2025-05-16:
I have implemented a first version of the enhanced multi-DEF reader. Using the convoluted example from issue #2014 (and follow-up private communication) gave me a slightly smaller (by 0.3%) OAS. Reading is somewhat faster (8s compared to 12s). XOR is clean.

The fix was not really easy, but maybe there are use cases for which the effort is more worth the benefit. I hope the patch isn't breaking anything. I don't have many multi-DEF test cases.

Matthias

stefanottili, 2025-05-17:
The fix created a dbdiff'erence because it didn't uniquify a via that exists in two def files with the same name but different geometry.
Minimally working testcase reduced from https://github.com/efabless/caravel sent via e-mail.

stefanottili, 2025-05-17:
The full caravel collects reads 13 def files, and that's without the additional user_wrapper def's.

Reading all ~240'is lef/gds files only once makes quite a runtime difference even for this tiny chip.

klayoutmatthias, 2025-05-17:
> The fix created a dbdiff'erence because it didn't uniquify a via that exists in two def files with the same name but different geometry. Minimally working testcase reduced from https://github.com/efabless/caravel sent via e-mail.

Thanks for this observation! I was afraid of such a case :(

Let's see if there is a simple fix for this. I assume that via caching needs to be globalized.

closed 2025-05-29 by klayoutmatthias