KLayout/klayout #2043: segfault with attached test case

oharboe, 2025-05-08:
untar [klayout-crash.tar.gz](https://github.com/user-attachments/files/20101053/klayout-crash.tar.gz)

Run

```
klayout -zz -rd design_name=lb_32x128 \
        -rd in_def=.//results/asap7/lb_32x128/base/6_final.def \
        -rd in_files="external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/platforms/asap7/gds/asap7sc7p5t_28_R_220121a.gds  " \
        -rd seal_file="" \
        -rd out_file=.//results/asap7/lb_32x128/base/6_1_merged.gds \
        -rd tech_file=.//objects/asap7/lb_32x128/base/klayout.lyt \
        -rd layer_map= \
        -r external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/util/def2stream.py
```

Output:

```
[INFO] Reporting cells prior to loading DEF ...
	external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/platforms/asap7/gds/asap7sc7p5t_28_R_220121a.gds
ERROR: Signal number: 11
Address: 0xa9
Program Version: KLayout 0.29.1 (2025-02-22 rLatestSourcePackage)

Backtrace:
/usr/lib/klayout/libklayout_lay.so.0 +0x2aa65b lay::enable_signal_handler_gui(bool) [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x45810 __restore_rt [libc_sigaction.c:?]
/lib/x86_64-linux-gnu/libpython3.13.so.1.0 +0x283a08 PyErr_Restore [??:?]
/lib/x86_64-linux-gnu/libpython3.13.so.1.0 +0x165710 PyObject_VisitManagedDict [??:?]
/lib/x86_64-linux-gnu/libpython3.13.so.1.0 +0x19b0e2 PyType_GenericAlloc [??:?]
/lib/x86_64-linux-gnu/libpython3.13.so.1.0 +0x28c836 PyErr_FormatUnraisable [??:?]
/lib/x86_64-linux-gnu/libpython3.13.so.1.0 +0x28df9e PyGC_Collect [??:?]
/lib/x86_64-linux-gnu/libpython3.13.so.1.0 +0x404e2b _Py_Get_Getpath_CodeObject [??:?]
/usr/lib/klayout/libklayout_pya.so.0 +0x4ea95 pya::PythonInterpreter::~PythonInterpreter() [??:?]
/usr/lib/klayout/libklayout_pya.so.0 +0x4eab5 pya::PythonInterpreter::~PythonInterpreter() [??:?]
/usr/lib/klayout/libklayout_lay.so.0 +0x15248f lay::ApplicationBase::shutdown() [??:?]
/usr/lib/klayout/libklayout_lay.so.0 +0x161085 lay::NonGuiApplication::~NonGuiApplication() [??:?]
/usr/lib/klayout/libklayout_lay.so.0 +0x161139 non-virtual thunk to lay::NonGuiApplication::~NonGuiApplication() [??:?]
/usr/lib/klayout/klayout +0x3b99 ?? [??:0]
/usr/lib/klayout/libklayout_rba.so.0 +0x2a1f4 rba::RubyInterpreter::eval_string(char const*, char const*, int, int) [??:?]
/lib/x86_64-linux-gnu/libruby-3.3.so.3.3 +0x2f53ba rb_error_arity [??:?]
/lib/x86_64-linux-gnu/libruby-3.3.so.3.3 +0x307e74 rb_check_funcall [??:?]
/lib/x86_64-linux-gnu/libruby-3.3.so.3.3 +0x30e851 rb_check_funcall [??:?]
/lib/x86_64-linux-gnu/libruby-3.3.so.3.3 +0x1075ec rb_check_copyable [??:?]
/lib/x86_64-linux-gnu/libruby-3.3.so.3.3 +0x10b173 ruby_run_node [??:?]
/usr/lib/klayout/libklayout_rba.so.0 +0x2ae9d rba::RubyInterpreter::initialize(int&, char**, int (*)(int&, char**)) [??:?]
/usr/lib/klayout/klayout +0x43df ?? [??:0]
/lib/x86_64-linux-gnu/libc.so.6 +0x2a338 __libc_start_call_main [libc_start_call_main.h:74]
/lib/x86_64-linux-gnu/libc.so.6 +0x2a3fb call_init [libc-start.c:128]
/usr/lib/klayout/klayout +0x4ca5 ?? [??:0]

Crash log written to /home/oyvind/.klayout/klayout_crash.log
```

stefanottili, 2025-05-08:
```
run-me-lb_32x128-asap7-base.sh
OpenROAD v2.0-21321-gc7d97cc3dc 
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Error: final_report.tcl, 3 Traceback (most recent call last):
  File "klayout_crash/external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/scripts/non_stage_variables.py", line 10, in <module>
    import yaml
ModuleNotFoundError: No module named 'yaml'
```

'find * | grep "\.gds"' shows that the testcase tar file doesn't contain the leaf cell gds

Instead of trying to debug the combination of shell variables and def2stram.py:
Here's how to use klayout's command line buddy tool strm2oas instead of klayout + def2stream.py
```
make
strm2oas \
	\
	--lefdef-map=\
	~/asap7_pdk_r1p7/cdslib/setup/asap7_fromAPR_08b.map \
	\
	--lefdef-lefs=\
	./objects/asap7/lb_32x128/base/klayout_tech.lef,\
	./objects/asap7/lb_32x128/base/asap7sc7p5t_28_R_1x_220121a.lef \
	\
	--lefdef-lef-layouts=\
	~/OpenROAD-flow-scripts/flow/platforms/asap7/gds/asap7sc7p5t_28_R_220121a.gds \
	\
	./results/asap7/lb_32x128/base/6_final.def \
	\
	6_final.oas
Warning: DBU of macro layout file '~/OpenROAD-flow-scripts/flow/platforms/asap7/gds/asap7sc7p5t_28_R_220121a.gds' does not match reader DBU (layout DBU is 0.00025, reader DBU is set to 0.001) (file=6_final.def)
```

oharboe, 2025-05-08:
Thanks! I did try to build klayout on my machine, ubuntu 25.04(not my choice, it is forced on me by other projects), but it failed with some qt build errors.

stefanottili, 2025-05-08:
I'm building openroad and klayout on a MacBook M1 using homebrew (works with both qt5 and qt6) and Windows WSL default ubuntu.

stefanottili, 2025-05-08:
```ERROR: RuntimeError: Duplicate MACRO name: A2O1A1Ixp33_ASAP7_75t_R (inside MACRO) (line=42, cell=,file=asap7sc7p5t_28_R_1x_220121a.lef) in Layout.read```
indicates that you read some lef file twice.

Later versions of klayout import->DEF/LEF read every filename only once, even if specified multiple times.

[image]

If this option is set, klayout silently reads all lef files in the same directory as the def file.
This lead to "Duplicate MACRO name" without feedback what was causing it.

stefanottili, 2025-05-08:
The issue with duplicate MACRO due to reading the same lef file multiple times got fixed in 0.29.2 Release date: 2024-06-06

https://github.com/KLayout/klayout/issues/1724

stefanottili, 2025-05-08:
to avoid reading lef files that happen to be in the same directory as def file use the strm2oas option --lefdef-no-implicit-lef

klayoutmatthias, 2025-05-08:
Hi @oharboe,

is there a reason you are using a specific version? 

Python 3.13 is pretty recent, so it's unlikely an old version like 0.29.1 would work properly with this Python version. My best guess is 0.30.1 (the latest), which is a pretty stable and useful version. If you want to repeat the build experiment, please try with that version. Also, 0.30.1 is compatible with Qt 6.9.

BTW: I have no experience with Ubuntu 25 yet. I also think it is not wise to use that version as the odd ones are not necessarily stable. I prefer using LTS versions such as 24.04. However, I don't think that the problems are related to the Ubunut version, but rather to the recent Qt and/or Python versions.

Matthias