KLayout/klayout #2048: LVS: Series resistance reduction causes floating resistor.

d-m-bailey, 2025-05-13:
Using klayout version `0.30.1`

The following extracted resistors
```
* device instance $78 r0 *1 -36.02,68.345 rppd
R$78 vss \$76 rppd w=3u l=38.65u ps=0 b=0 m=1            * \$76 incorrectly changed to \$78 when reduced
* device instance $79 r0 *1 -36.02,72.165 rppd
R$79 \$78 \$76 rppd w=0.5u l=38.65u ps=0 b=0 m=1         * This resistor becomes floating
* device instance $80 r0 *1 -36.02,73.485 rppd           * vvvvvvvvvvvvvvvvv
R$80 \$78 \$80 rppd w=0.5u l=38.65u ps=0 b=0 m=1         * 3 resistors reduced to R$82 below 
* device instance $81 r0 *1 -36.02,74.805 rppd
R$81 \$84 \$80 rppd w=0.5u l=38.65u ps=0 b=0 m=1
* device instance $82 r0 *1 -36.02,76.125 rppd
R$82 \$84 \$44 rppd w=0.5u l=38.65u ps=0 b=0 m=1         * ^^^^^^^^^^^^^^^
* device instance $83 r0 *1 -36.02,77.445 rppd           * vvvvvvvvvvvvvvvvv
R$83 \$88 \$44 rppd w=0.5u l=38.65u ps=0 b=0 m=1         * 5 resistors reduced to R$87 below
* device instance $84 r0 *1 -36.02,78.765 rppd                   
R$84 \$88 \$93 rppd w=0.5u l=38.65u ps=0 b=0 m=1
* device instance $85 r0 *1 -36.02,80.085 rppd
R$85 \$96 \$93 rppd w=0.5u l=38.65u ps=0 b=0 m=1
* device instance $86 r0 *1 -36.02,81.405 rppd
R$86 \$96 \$99 rppd w=0.5u l=38.65u ps=0 b=0 m=1
* device instance $87 r0 *1 -36.02,82.725 rppd
R$87 \$73 \$99 rppd w=0.5u l=38.65u ps=0 b=0 m=1         * ^^^^^^^^^^^^^^^
```

are reduced to
```
* device instance $78 r0 *1 -36.02,68.345 rppd
R$78 vss \$78 rppd w=3u l=38.65u ps=0 b=0 m=1
* device instance $79 r0 *1 -36.02,72.165 rppd
R$79 \$78 \$76 rppd w=0.5u l=38.65u ps=0 b=0 m=1         * \$76 is not connected to anything in the extracted netlist.
* device instance $82 r0 *1 -36.02,76.125 rppd
R$82 \$78 \$44 rppd w=0.5u l=115.95u ps=0 b=0 m=1
* device instance $87 r0 *1 -36.02,82.725 rppd
R$87 \$73 \$44 rppd w=0.5u l=193.25u ps=0 b=0 m=1
```

To reproduce, run [this notebook](https://github.com/d-m-bailey/ihp-mpw-be/blob/klayout-reduce/TO_202504.ipynb) in colab.
Total execution time is less than 10 minutes.

klayoutmatthias, 2025-05-13:
Hi @d-m-bailey,

I have no idea how to use your notebook. If you want my support, please don't dump such a thing on me. Give me a klayout call, the layout and the schematic netlist. Then we're talking things I can reproduce, without wasting my night time with reproducing the problem.

Apparently we're talking about the bandgap_reference project. And apparently we're talking about a netlist with many R's. But the Spice dump (`[14]`) I see only a few resistors, not as many as in your dump above.

So maybe we're talking an extracted netlist here? In that case, KLayout is out. It does not do extraction.

Matthias

d-m-bailey, 2025-05-13:
@klayoutmatthias Thanks for the quick response. Sorry for dumping the colab notebook on you, but it was the easiest way for me to submit a reproducible result. The linked github page has a blue button at the top labeled `Open in Colab`. After that, just choose `Runtime` -> `Run all` from the menu. It will generate and display 2 netlists - one extracted with the `combine_devices` option and one extracted without.

In the original post, the first snippet is from the netlist extracted without `combine_devices`. The second snippet is from the netlist extracted with `combine_devices`. 

This is the `bandgap_ref_cmos` design from the April 2025 iHP shuttle [located here](https://github.com/IHP-GmbH/TO_Apr2025/tree/main/bandgap_ref_cmos). 

Are you able to execute Klayout with the ihp_sg13g2 `run_lvs.py` script from the open pdk? It simplifies things if you can because it references several files that I'd have to manage for a standalone result. 

> So maybe we're talking an extracted netlist here? In that case, KLayout is out. It does not do extraction.

Right, the reduced extracted netlist. I thought layout extraction and LVS were built into Klayout. Can I ask what you mean by "It doesn't do extraction"?

Based on your response to the questions above, I will try to create a standalone test case.

klayoutmatthias, 2025-05-14:
That is what I mean - the notebook is too convoluted to act as a test case.

KLayout only does LVS. That is a functional check against the netlist written from the schematic. No parasitics involved. We are working on that, but that is not yet available.

I assume, parasitic extraction is done in Magic. Also, I don't think that netlist reduction is done in KLayout. There is some very basic features, but I can't imagine it is used here. 

So bottom line is, I don't think KLayout is involved her, but in order to find out, I had to understand the notebook.

Matthias

d-m-bailey, 2025-05-14:
@klayoutmatthias Ok, I think I understand where my explanation was lacking. Thanks for your patience.

The layout in question _does not_ have parasitic resistors. These are explicit poly resistors. The problem that I think I'm seeing is that the netlist extracted with `combined_devices` enabled is not equivalent to the netlist extracted without `combined_devices`.

With `combined_devices` disabled, I get these resistors.
```
* device instance $78 r0 *1 -36.02,68.345 rppd
R$78 vss \$76 rppd w=3u l=38.65u ps=0 b=0 m=1
* device instance $79 r0 *1 -36.02,72.165 rppd
R$79 \$78 \$76 rppd w=0.5u l=38.65u ps=0 b=0 m=1
* device instance $80 r0 *1 -36.02,73.485 rppd
R$80 \$78 \$80 rppd w=0.5u l=38.65u ps=0 b=0 m=1
* device instance $81 r0 *1 -36.02,74.805 rppd
R$81 \$84 \$80 rppd w=0.5u l=38.65u ps=0 b=0 m=1
* device instance $82 r0 *1 -36.02,76.125 rppd
R$82 \$84 \$44 rppd w=0.5u l=38.65u ps=0 b=0 m=1
* device instance $83 r0 *1 -36.02,77.445 rppd
R$83 \$88 \$44 rppd w=0.5u l=38.65u ps=0 b=0 m=1
* device instance $84 r0 *1 -36.02,78.765 rppd
R$84 \$88 \$93 rppd w=0.5u l=38.65u ps=0 b=0 m=1
* device instance $85 r0 *1 -36.02,80.085 rppd
R$85 \$96 \$93 rppd w=0.5u l=38.65u ps=0 b=0 m=1
* device instance $86 r0 *1 -36.02,81.405 rppd
R$86 \$96 \$99 rppd w=0.5u l=38.65u ps=0 b=0 m=1
* device instance $87 r0 *1 -36.02,82.725 rppd
R$87 \$73 \$99 rppd w=0.5u l=38.65u ps=0 b=0 m=1
* device instance $88 r0 *1 -36.015,84.045 rppd
R$88 VBG \$104 rppd w=0.71u l=38.65u ps=0 b=0 m=1
* device instance $89 r0 *1 -36.015,85.575 rppd
R$89 \$116 \$104 rppd w=0.71u l=38.65u ps=0 b=0 m=1
* device instance $90 r0 *1 -36.015,87.105 rppd
R$90 \$116 \$117 rppd w=0.71u l=38.65u ps=0 b=0 m=1
* device instance $91 r0 *1 -36.015,88.635 rppd
R$91 \$125 \$117 rppd w=0.71u l=38.65u ps=0 b=0 m=1
* device instance $92 r0 *1 -36.015,90.165 rppd
R$92 \$125 \$126 rppd w=0.71u l=38.65u ps=0 b=0 m=1
* device instance $93 r0 *1 -36.015,91.7 rppd
R$93 \$140 \$126 rppd w=0.71u l=38.65u ps=0 b=0 m=1
* device instance $94 r0 *1 -36.015,93.23 rppd
R$94 \$140 vss rppd w=0.71u l=38.65u ps=0 b=0 m=1
```
This corresponds to this layout where each individual resistor is 38.65um long.

[image]
Note: the `¥` mark should be `\` in the screen shot annotation.

Now in the netlist extracted with `combine_devices` enabled, I get these resistors.
```
* device instance $78 r0 *1 -36.02,68.345 rppd
R$78 vss \$78 rppd w=3u l=38.65u ps=0 b=0 m=1
* device instance $79 r0 *1 -36.02,72.165 rppd
R$79 \$78 \$76 rppd w=0.5u l=38.65u ps=0 b=0 m=1
* device instance $82 r0 *1 -36.02,76.125 rppd
R$82 \$78 \$44 rppd w=0.5u l=115.95u ps=0 b=0 m=1
* device instance $87 r0 *1 -36.02,82.725 rppd
R$87 \$73 \$44 rppd w=0.5u l=193.25u ps=0 b=0 m=1
* device instance $94 r0 *1 -36.015,93.23 rppd
R$94 VBG vss rppd w=0.71u l=270.55u ps=0 b=0 m=1
```
`R$78`, and `R$79` are the same individual resistors as before while `R$82` is the combination of `R$80`, `R$81`, and `R$82`. I expected `R$79` to be included in this resistor chain, but it is not.
The original ` \$76` connection of `R$78` has been switched to ` \$78`, which leaves `R$79` with an unconnected ` \$76`.

Here's the layout. 

[image]

If you have access to the ihp-sg130g2 open pdk, I can send the commands for a test case. If you don't, I have to create a klayout rule file because the pdk references many individual files.

klayoutmatthias, 2025-05-14:
Hi @d-m-bailey,

I see we are getting closer.

I have access to the PDK, but a CDL netlist is helpful, so I do not need to netlist the schematic.

The LVS is heavily customized as far as I recall. The R devices are not the native ones as they do not come with a computer value. So maybe the customization plays a role here too.

Matthias

klayoutmatthias, 2025-05-14:
Maybe I can take the Spice netlist from the log instead of CDL. I will try.

Matthias

klayoutmatthias, 2025-05-14:
Okay, this is how I can reproduce it without much installation effort:

* Clone IHP PDK from https://github.com/IHP-GmbH/IHP-Open-PDK

* Get Layout from https://github.com/IHP-GmbH/TO_Apr2025/blob/main/bandgap_ref_cmos/design_data/gds/FMD_QNC_15_WeakInvBGR.gds

* Pasted the SPICE netlist from the notebook logs int "bandgap_ref_cmos.spice"

* Run the LVS without device combination:

```
python3 <path-to-PDK>/ihp-sg13g2/libs.tech/klayout/tech/lvs/run_lvs.py --layout=FMD_QNC_15_WeakInvBGR.gds --netlist=bandgap_ref_cmos.spice --run_dir=lvs_run_nodc --no_series_res --no_parallel_res
```

* Run the LVS with device combination:

```
python3 <path-to-PDK>/ihp-sg13g2/libs.tech/klayout/tech/lvs/run_lvs.py --layout=FMD_QNC_15_WeakInvBGR.gds --netlist=bandgap_ref_cmos.spice --run_dir=lvs_run_withdc
```

Now I can start debugging ...

Matthias

d-m-bailey, 2025-05-14:
@klayoutmatthias I'm really thankful for your patience and persistence in looking into this issue. (I'm a big fan of Klayout.)

klayoutmatthias, 2025-05-14:
Thanks ... :)

I think I have found the problem. I cannot get LVS to match yet, because the SPICE file isn't a full CDL (for example, the tap diodes are not there). But I can get a proper netlist without floating nets.

The root cause is in the customization in the IHP PDK I think. There is a logic error.

In `custom_combiner.lvs', module `DeviceCombinerMethods` they define a method `supp_series` which is probably supposed to check where two devices can be combined in a serial fashion. This method is used inside `RESDeviceCombiner` like this:

```
    # Check if terminals series or parallel (With same params)
    if supp_parallel(a, b, 0, 1)
      return false unless PARALLEL_RES
      return false unless %w[w l ps b].all? { |param| same_parameter(a, b, param) }

      a.set_parameter('m', a.parameter('m') + b.parameter('m'))
    elsif supp_series(a, b, 0, 1)
      return false unless SERIES_RES
      return false unless %w[w ps b m].all? { |param| same_parameter(a, b, param) }

      a.set_parameter('l', a.parameter('l') + b.parameter('l'))
    else
      return false
    end
```

Problem is: when "supp_series" returns true, it has already changed the connection of the device. In case of two resistors with "w=3" and "w=0.5" the "same_parameter" check rejects the serial combination. In that case, the resistors are already disconnected even though the serial combination is rejected. That causes the floating net.

Fix is to separate the check and the actual reconnection:

```
  # A helper function to check whether two device connected in series
  def supp_series(a, b, net1, net2)
    a_net1 = a.net_for_terminal(net1)
    a_net2 = a.net_for_terminal(net2)
    b_net1 = b.net_for_terminal(net1)
    b_net2 = b.net_for_terminal(net2)

    return false unless a_net1 && b_net1 && a_net2 && b_net2

    series_con_a1 = a_net1.expanded_name == b_net1.expanded_name || a_net1.expanded_name == b_net2.expanded_name
    series_con_a2 = a_net2.expanded_name == b_net1.expanded_name || a_net2.expanded_name == b_net2.expanded_name
    series_con_a1 || series_con_a2

    # NOTE: no connection changed

  end

  # NEW method
  def reconnect_series(a, b, net1, net2)
    a_net1 = a.net_for_terminal(net1)
    a_net2 = a.net_for_terminal(net2)
    b_net1 = b.net_for_terminal(net1)
    b_net2 = b.net_for_terminal(net2)

    series_con_a1 = a_net1.expanded_name == b_net1.expanded_name || a_net1.expanded_name == b_net2.expanded_name
    series_con_a2 = a_net2.expanded_name == b_net1.expanded_name || a_net2.expanded_name == b_net2.expanded_name

    # Reroute terminal connections based on cluster IDs
    if series_con_a1
      if a_net1.expanded_name == b_net1.expanded_name
        a.connect_terminal(0, b_net2)
      else
        a.connect_terminal(0, b_net1)
      end
    end

    if series_con_a2
      if a_net2.expanded_name == b_net1.expanded_name
        a.connect_terminal(1, b_net2)
      else
        a.connect_terminal(1, b_net1)
      end
    end

  end

....
Later:

    # Check if terminals series or parallel (With same params)
    if supp_parallel(a, b, 0, 1)
      return false unless PARALLEL_RES
      return false unless %w[w l ps b].all? { |param| same_parameter(a, b, param) }

      a.set_parameter('m', a.parameter('m') + b.parameter('m'))
    elsif supp_series(a, b, 0, 1)
      return false unless SERIES_RES
      return false unless %w[w ps b m].all? { |param| same_parameter(a, b, param) }

      a.set_parameter('l', a.parameter('l') + b.parameter('l'))
      # NEW make the connection after confirming the devices can be combined
      reconnect_series(a, b, 0, 1)
    else
      return false
    end

    # Disconnect the second device and let the system clean it up
    res_nets.each do |term| 
      b.disconnect_terminal(term) 
    end

...
    true
  end
```

This is a pretty basic change, so I cannot give a small patch.

There is a lot to say here regarding performance and style. I parts, the LVS is overly complex, because the API I opened is pretty sparse and needs a lot of customization. I am open to a discussion how this overhead can be reduced. For example, there is a nice base class for a two-terminal combiner that just isn't exposed.

There are some further issues that probably have a severe performance impact, but I did not check the full implementation. I guess that there is some improvement potential here.

In any case, I think the problem is not inside KLayout, but in the IHP PDK customization.

Matthias

d-m-bailey, 2025-05-15:
@klayoutmatthias Thanks so much for debugging this. I now understand the condition that creates the disconnect.

I'll forward this result to the rule maintainers while working on a fix myself.

The necessity and appropriateness of tap diodes (netlisted as resistors) are a slight issue of contention at the moment. 

I disabled the extraction with this
```
sed -i -e '/^well_patt/s/"/"-/' -e '/^sub_patt/s/"/"-/' $PDK_ROOT/$PDK/libs.tech/klayout/tech/lvs/rule_decks/general_derivations.lvs
```
This changes 
```
# n1/p1 taps labels
well_patt = glob_to_case_insensitive_glob("well")
sub_patt = glob_to_case_insensitive_glob("sub!")
```
to
```
# n1/p1 taps labels
well_patt = glob_to_case_insensitive_glob("-well")
sub_patt = glob_to_case_insensitive_glob("-sub!")
```
and since these labels do not exist, the wells are not tagged and n/ptaps are not extracted.

closed 2025-05-15 by d-m-bailey