KLayout/klayout #2051: L2N from a RecursiveShapeIterator  from layout of a library cannot find subcircuit pin names

sebastian-goeldi, 2025-05-16:
Hi Matthias,

I know sounds like a stupid convoluted request, but I think nonetheless worth looking at (though low prio). I am using layouts in kfactory which are always owned by a Library. I tried to implement some basic electrical LVS and the pin names were always empty strings. I will prepare a minimal example tomorrow.

I in the end fixed it by `library.layout.dup()` to get an independent layout object.

Expected Behavior:

Either complain if the original layout has a `.library()` or fix the recursive shape iterator (or `extract_netlist()`) to properly pick up names.

klayoutmatthias, 2025-05-16:
Hi Sebastian,

I am looking forward to a test case :)

Actually, there is no real reason that this shouldn't work. Maybe it's just a text translation issue, but definitely a bug.

Matthias

sebastian-goeldi, 2025-05-17:
Unfortunately I cannot reproduce it in a standalone context. I am very confused now.

I checked whether it is because one of my objects is pointing to the wrong layout or index or similar. If you want, I can post what I believed was the error:

```python
import klayout.db as kdb

# working example

ly = kdb.Layout()
ly.read("~/Downloads/extraction_test.gds.gz")

tc = ly.top_cell()
l2n = kdb.LayoutToNetlist(kdb.RecursiveShapeIterator(ly, tc, []))

m1 = l2n.make_layer(ly.layer(2, 0), "METAL1")
m2 = l2n.make_layer(ly.layer(4, 0), "METAL2")
v1 = l2n.make_layer(ly.layer(3, 0), "VIA1")

l2n.connect(m1)
l2n.connect(m2)
l2n.connect(m1, v1)
l2n.connect(v1, m2)

l2n.extract_netlist()

l2n.write("l2n_layout.l2n", True)

# library example

lib = kdb.Library()

lib.register("EXTRACTION_LIB")
ly = lib.layout()
ly.read("extraction_test.gds.gz")

lib_dup = lib.dup()
lib_dup.register("EXTRACTION_LIB_DUP")
ly = lib_dup.layout()

tc = ly.top_cell()
l2n = kdb.LayoutToNetlist(kdb.RecursiveShapeIterator(ly, tc, []))

m1 = l2n.make_layer(ly.layer(2, 0), "METAL1")
m2 = l2n.make_layer(ly.layer(4, 0), "METAL2")
v1 = l2n.make_layer(ly.layer(3, 0), "VIA1")

l2n.connect(m1)
l2n.connect(m2)
l2n.connect(m1, v1)
l2n.connect(v1, m2)

l2n.extract_netlist()

l2n.write("l2n_library.l2n", True)
```

[extraction_test.gds.gz](https://github.com/user-attachments/files/20263963/extraction_test.gds.gz)

I can reproduce the error within kfactory, but not in standalone mode.

Sorry for the trouble. I am ok to just use the way I did it now. I don't think there's a big problem in that. I am now just copying the layout of the library directly and then it works within kfactory. I do not fully grasp what the difference is. Maybe some internal object is still pointing to the wrong layout and therefore it can't find the shapes correctly. (Though the pins do seem to exist, the names are just empty strings)

closed 2025-05-17 by sebastian-goeldi

klayoutmatthias, 2025-05-17:
Hi @sebastian-goeldi,

If you like, you can paste a kfactory code example which allows reproducing the problem.

Matthias

sebastian-goeldi, 2025-05-26:
Hi @klayoutmatthias ,

I think I finally found my error. I think it was related to getting layer indexes and their infos. And I think in some cases this chain caused layers to only be defined by name and if they were not properly registered, the resulting layer index would be non-existant (and I believe mapped to -1).

 Thanks again for your help!

klayoutmatthias, 2025-05-26:
I didn't do anything, but you are welcome :)