KLayout/klayout #2054: arm wheels for linux

joamatab, 2025-05-19:
How can we get linux arm wheels?

[image]

Maybe also windows arm wheels

klayoutmatthias, 2025-05-20:
Hi @joamatab,

GitHub wanted to offer ARM runners, but I am not sure if they are available already.

Regarding Windows, these builds are running on Azure.  Do they offer ARM runners?

If both do, providing the wheels should not be an issue. 

Matthias

sebastian-goeldi, 2025-05-21:
Gh arm seems to be available as public preview (at the very least, maybe also silently released).

Should be available as `ubuntu-24.04-arm`
(Or potentially `ubuntu-latest-arm` as well)

Same for windows `windows-11-arm`

As gh runners run on azure, I am pretty sure all of it is available in azure as well. Not sure about the azure setup you use.

tvt173, 2025-05-21:
yes, `ubuntu-24.04-arm` should work. i've done this for a few other packages recently... i just made a PR for this #2058