KLayout/klayout #2057: feature request: nanometer scalebar

Fusion86, 2025-05-20:
It looks like the scalebar/ruler always shows the scale in micrometers. For more modern tech nodes it would be nice to have a nanometer scale. This would probably only require a text formatting change.

Fusion86, 2025-05-22:
A sorry, I should've been more specific. What I meant is that it'd be nice to print the sizes in nanometers and not micrometers. So 40nm instead of 0.040um.
Honestly this is just more of a visual thing, technically it is exactly the same. Printing things in nanometers (when appropriate) just signals to the user "yes we support these smaller sizes".

klayoutmatthias, 2025-05-25:
I am switching to nm units now if the value is below 0.1µm

[image]

Correspondingly it will switch to mm above 100µm and even to meters further up. I did not include km or Parsecs, as I think that meters should be good enough for a full wafer.

Matthias

closed 2025-05-26 by klayoutmatthias