KLayout/klayout #2059: How to contact the bulk of Nmos to the ground with Klayout

ElyasPuffu, 2025-05-22:
Hi,

I’m working on designing an NMOS transistor layout in KLayout, and I want to make sure the bulk (substrate) is properly connected to ground (GND). I

Could anyone please explain or point me to a resource showing how to properly implement this in KLayout? Specifically:
        •        What layers should I use for the P+ diffusion and contact?
        •        How do I connect it to the metal layer for GND?
        •        Are there any predefined cells or best practices I should follow?

Any help or example would be greatly appreciated. Thanks in advance!

ElyasPuffu, 2025-05-22:
[image]

klayoutmatthias, 2025-05-22:
I think this is an AI generated "issue".