KLayout/klayout #2063: Missing polygon

alexeys85, 2025-05-25:
I'm trying to view some small [GDS2 file](https://www.yzuda.org/download/_GDSII_examples/inv.gds2).
gdstk shows that layers 49 and 51 contain small polygon with vertices at: (-2.0, -2.0) (2.0, -2.0) (2.0, 2.0) (-2.0, 2.0).
However KLayout doesn't show this rectangle.

Kazzz-S, 2025-05-25:
Those polygons seem to be components of the cell `via`.
Then, the cell `via` is SREFed twice by the cell `inv1` at different locations (not (0,0)), and the results are displayed as expected.

[image]

[image]

stefanottili, 2025-05-25:
Ha, Kazzz just beat me to the punch ...

> However KLayout doesn't show this rectangle.

It does for me when using the hotkey `*`, or the Menu Item `Display->Full Hierarchy`.
Or use the mouse to set the `Levels` as shown at the bottom of the cells window to from 0 down to level 2.