KLayout/klayout #2068: Setting correspondence between internal nets

d-m-bailey, 2025-05-29:
Note: Copied from [this question](https://matrix.to/#/!pbFlihdmeopjylaJqj:fossi-chat.org/$N9Bz_FLjqRR78JUIaVyBs0vhAGkFrn5qL-8kFOxka4w?via=fossi-chat.org&via=matrix.org&via=underzsea.com) on element.fossi-chat.org

A question about klayout LVS cross reference output. The extracted netlist contains internal net names like `\$I12` that are output in the log if there is a mismatch.

[image]

However, in the cross reference, the extracted nets have the form `$30`.

[image]

How does one determine the correspondence between (or set equivalency, i.e. `same_nets`, for) the extracted net names `\$I12` and the layout nets in the cross reference `$30`?

klayoutmatthias, 2025-05-29:
Hi @d-m-bailey,

Internal nets cannot be cross-referenced. You need to name a net (label it) before you can do so.

The internal net names are arbitrary numbers and they may even change between runs. 

BTW: there is an issue that is fixed with the upcoming 0.30.2 release: the internal net names are renumbered when the LVS DB is loaded - the internal net names reported in the log messages don't match the ones from the cross reference view and the generated netlist.

The new version will also have the "flag_missing_ports" feature.

Matthias

d-m-bailey, 2025-05-29:
@klayoutmatthias As always, thanks for your prompt response!

The reason I wanted to establish correspondence points was to try to pinpoint the discrepancy in an LVS result. Currently, I'm using `same_nets!("*", "*")` to force correspondence to the labeled nets. If you have time, could you take a look at it and offer some advice? Here's a tarball with the source netlist, extracted netlist, gds, and lvsdb files. It's [this ihp-sg13g2 design](https://github.com/IHP-GmbH/TO_Apr2025/tree/main/160GHz_LNA) from the April 2025 shuttle. I've made some modifications to the rule files, but I'm thinking you shouldn't need those to find the discrepancy.

[160GHz_LNA.tgz](https://github.com/user-attachments/files/20512276/160GHz_LNA.tgz)

If it's not immediately obvious, please don't spend too much time on it, but any tips you could give on how to pinpoint discrepancies would be greatly appreciated. Maybe labeling the normally internal nets to establish correspondence?

klayoutmatthias, 2025-05-30:
What should I say ... despite your warning, I spent some time debugging to make sure there isn't a bug on my side.

I used main HEAD from the IHP PDK (rev. 68eebafcd9b2f5e92c69d37a8d3d90eb266550f5) and honestly I don't know how RFCMIM could every work.

Problem #1: the device name is wrong (should be "cap_rfcmim" in line 38 of `cap_extraction.lvs` instead of "rfcmim"). Otherwise it won't with your CDL.

Problem #2: Cap "RC3" lacks the "u" unit for "w" and the Spice reader customization implies "meters" in that case. So I changed:

```
CRC3 net2_3 net3 GND cap_rfcmim w=5.2 l=2.4u
```

in the CDL to:

```
CRC3 net2_3 net3 GND cap_rfcmim w=5.2u l=2.4u
```

Problem #3: the "wfeed" parameter was declared as a primary one, hence it was compared. As it is not set in the schematic, this is an error. I don't know if it is intended to be extracted for compare. I assume this is not the case, so I changed it to "non-primary":

This Line in "custom_mim_extractor.lvs"

```
    add_parameter(RBA::DeviceParameterDefinition.new('wfeed', 'feed width', 0, true))
```

becomes

```
    add_parameter(RBA::DeviceParameterDefinition.new('wfeed', 'feed width', 0, false))
```

Problem #4: The substrate rectangle is included in the device definition for "rfcmim". This makes *all* rfcmim combined into a single chunk. The device extractor complains with a warning therefore. 

The reason is the definition of the device cluster, which is done through a pseudo-connectivity. This should not include the big substrate rectangle. Instead - like for the other devices - the substrate is an output layer only where KLayout will place the substrate pins.

I removed this piece of code from "custom_mim_extractor.lvs":

```
    if name.downcase.include?('rfcmim')
      sub_mk = layers[5]
      conn.connect(sub_mk, dev_mk)
    end
```

and replaced this code

```
      if sub_mk.is_empty?
        $logger.info("Sub terminal for #{@name} device doesn't exist, please recheck")
        return nil
      else
        define_terminal(device, @reg_dev.terminal_id("mim_sub"), 5, sub_mk[0])
```

by this code

```
        define_terminal(device, @reg_dev.terminal_id("mim_sub"), 5, btm_mim[0])
```

this code will copy the bottom plate marker as terminal for the substrate.

With these changes, LVS matches in my case.

Matthias

klayoutmatthias, 2025-05-30:
Hi @d-m-bailey,

The `wfeed` parameter causes a mismatch, no warning. So it must be either disabled (not made primary), removed or added to the CDL.

I don't fully understand how oversizing `pwell_block` would solve the issue. But I think that substrate should not go into the device formation - at least not as a large rectangle joining multiple devices. After all, the device geometry is not defined by substrate. If you see it as an output only, where the substrate terminals are laid out, it will serve the purpose.

Matthias

klayoutmatthias, 2025-05-31:
Yes, exactly. I decided for "typed" devices - i.e. parameters are defined by the device class and are always present for a device with a given class. Contrary to parameters as a loose dictionary of key/values. I felt this is more consistent - for example the serial and parallel combination of devices changes the parameter values in device-class specific ways. So the algorithms need to rely on the fact that certain parameters are present and are defined in a specific way.

The IHP LVS heavily customizes the devices - maybe too much. If things are inconsistent, you may observe strange behavior. 

I found another example where they are not quite consistent: the TOPMETAL resistors use a device class called "RES2" in layout extraction, which disables the "R" parameter and enables "W" and "L".

Inside the SPICE reader, they generate a standard resistor class, which uses "R" as primary and "W" and "L" as secondary (disabled) parameters.

So depending on the perspective, you should compare "R" or "W" and "L". It is much cleaner to use "RES2" also for the schematic reader.

BTW: I came across that because I tried to make the parameter nature more clearer in the LVS cross reference view and placed the secondary ones inside brackets. Which gave me this:

[image]

Matthias

d-m-bailey, 2025-06-01:
@klayoutmatthias I have verified that adding the missing `wfeed` parameter and the unit `u` to `RC3` to the test cdl that I'm using results in a clean LVS result (with the modified extraction rules that I'm using).

So the `cap_rfcmim` devices were not matching because the source netlist was missing the `wfeed` parameter. The layout `cap_rfcmim[L,W,wfeed]` doesn't match the source `cap_rfcmim[L,W]`.

However, in your previous comment, there are the layout `res_topmetal2[L,W]` resistors match the source `top_resmetal2[R]` device types. I'm confused about why the `res_topmetal2` would match, but not the `cap_rfcmim`.

klayoutmatthias, 2025-06-01:
Yes I agree, that's a bit confusing 

But those are different cases: in the RFCMIM case the definition was consistent, but the parameters were wrong.

In the TOPMETAL case, there are inconsistent device models (classes) and it boiled doesn to having W and L as primary parameters. In this case they match. Had KLayout selected the schematic version, they would not match as in that case the primary parameter is R.

KLayout gives priority to the layout definition, but I am not sure if that is always the case.

Matthias

d-m-bailey, 2025-06-02:
@klayoutmatthias Thanks for your patient and detailed explanations.