KLayout/klayout #2072: Python 3.13.3, cannot import pya module

XavierFerland-Turbis, 2025-06-11:
Here's my error, thanks :)

Xavier

[image]

lukasc-ubc, 2025-06-11:
The documentation describes "pya".

I found the following:

KLayout on Python 3.11 uses

```
import klayout
import pya

```

"pya" is the same as import klayout.db, and it looks like "import pya" doesn't work in 3.13 (on Windows, on Mac OSX it works fine)

So instead, we need to use:

```
import klayout.db as pya
```

@klayoutmatthias is this correct?  Should the documentation be updated?

klayoutmatthias, 2025-06-11:
@lukasc-ubc  and @XavierFerland-Turbis,

It's also working with pya on 3.13 on Ubuntu. I don't have Windows available with 3.13, so I can't check or debug. Unless something is wrong with the installation in general, there should not be a reason "pya" isn't working. Basically it's just a redirection and depends only on pretty basic Python features.

But I'd recommend to stop using "pya" anyway. It's taken by another module (Python audio coding classes). Please use "klayout". Inside the application (KLayout UI), you can also use "klayout" now, so there is no reason to use "pya" any longer. I will eventually remove it from klayout to avoid confusion with the other module.

Matthias

lukasc-ubc, 2025-06-11:
Can you update your documentation web page?

Thank you