KLayout/klayout #2085: Why can't klayout open a project and only open a single file such as GDS

WenjunHuang94, 2025-07-07:
I hope Klayout can open a folder in the form of an engineering project. Can you support this feature

klayoutmatthias, 2025-07-07:
This is not an issue and lacks any proper description.

closed 2025-07-07 by klayoutmatthias

WenjunHuang94, 2025-08-22:
I have a project folder cell, which contains many GDS files. However, it seems that klayout can only select a single specific GDS file to open, and cannot open this folder cell. Can klayout support this operation? 

[image]

[image]

[image]