KLayout/klayout #2091: Open Recent takes a long time to load when it includes samba mounted files

simenmm, 2025-07-15:
Samba servers take a while to respond. When I've opened files on those servers and hover over File->Open Recent, all the servers are queried for the state of the file to see if it exists, and that causes KLayout to freeze for several seconds. Even accidentally moving the mouse over the open recent bar triggers this behavior. The more files the worse. Inserting the following line in .klayoutrc should demonstrate the issue.

<mru>'\\\\sambadrive\\example.gds'@'tech' </mru>

KLayout 0.30.2 on Windows 11.

closed 2025-07-15 by klayoutmatthias

simenmm, 2025-07-15:
I haven't seen this in any other software, so I must disagree that it's not a KLayout issue. If you open files on samba drives, the file menu becomes effectively unusable until you manually edit the config file. I can't count the number of times I had to kill KLayout from the task manager until I found the root cause. I believe that if file locations starting with \\\\\\\\ aren't checked automatically the problem will be resolved, since shares mounted with a drive letter don't seem to be affected.

klayoutmatthias, 2025-07-16:
You're invited to fix it then. It's Open Source.

simenmm, 2025-07-16:
I would but I'm not a software engineer. Not even sure where to start. But it sounds like the problem isn't correctly identified. I suspect it's unique to mounted network locations without a drive letter, and not to samba servers. Are you unable to replicate the behavior with the line I put in the issue?