KLayout/klayout #2094: Suspected performance issue in the python code to insert Regions into Shapes

engineeringvase, 2025-07-16:
In one line: When the Shapes class's insert function with a Region as argument, there seems to be an accumulating amount of time needed for subsequent executions. 

Put more plainly: First 
```
someShapes.insert(someRegion)
```
is very fast, after 1000 insertions it gets slower, after 2000 it gets slower still, ... I made a demo script to test this, see the script and the outcome below. The amount of time needed for the operation seems to increase linearly with the number times it has been executed in the past. For myself I found a workaround:  insert each part of the region separately:
```
for shape in someRegion.each():
    someShapes.insert(shape)
```
Note, I tried this with the python klayout.db installed through pip (yesterday), and also in the latest KLayout 0.30.2 through the Macro Developer, the exact same linear increase is observed in both cases.

### Demo script

```
import klayout.db as pya
import datetime

# Create layout, cell and layer
ly   = pya.Layout()
l1   = ly.layer(1,0)
main = ly.create_cell("Main")

# Demonstration: add tons of boxes
box1 = pya.Region(pya.Box(4000))
box2 = pya.Region(pya.Box(2000,8000))

print("")
print("## Problematic case")
t0 = datetime.datetime.now()
for i in range(1,20001):
    
    # Time keeping for demo
    if i%1000 == 0: 
        timecost = (datetime.datetime.now() - t0).total_seconds()
        print("Iteration", i, ", Time per 1000 iterations:", timecost)
        t0 = datetime.datetime.now() 
        
    # Problematic execution
    main.shapes(l1).insert(box1)
    main.shapes(l1).insert(box2)
    

print("")
print("## Alternative execution")
t0 = datetime.datetime.now()
for i in range(1,20001):
    
    # Time keeping for demo
    if i%1000 == 0: 
        timecost = (datetime.datetime.now() - t0).total_seconds()
        print("Iteration", i, ", Time per 1000 iterations:", timecost)
        t0 = datetime.datetime.now() 
        
    # Alternative execution
    for shape in box1.each():
        main.shapes(l1).insert(shape)    
    for shape in box2.each():
        main.shapes(l1).insert(shape)    

```

### Script outcome

```
## Problematic case
Iteration 1000 , Time per 1000 iterations: 0.05159
Iteration 2000 , Time per 1000 iterations: 0.107194
Iteration 3000 , Time per 1000 iterations: 0.16091
Iteration 4000 , Time per 1000 iterations: 0.212394
Iteration 5000 , Time per 1000 iterations: 0.267079
Iteration 6000 , Time per 1000 iterations: 0.320597
Iteration 7000 , Time per 1000 iterations: 0.374572
Iteration 8000 , Time per 1000 iterations: 0.426344
Iteration 9000 , Time per 1000 iterations: 0.482823
Iteration 10000 , Time per 1000 iterations: 0.530373
Iteration 11000 , Time per 1000 iterations: 0.578954
Iteration 12000 , Time per 1000 iterations: 0.635411
Iteration 13000 , Time per 1000 iterations: 0.694576
Iteration 14000 , Time per 1000 iterations: 0.747631
Iteration 15000 , Time per 1000 iterations: 0.799844
Iteration 16000 , Time per 1000 iterations: 0.847128
Iteration 17000 , Time per 1000 iterations: 0.932573
Iteration 18000 , Time per 1000 iterations: 0.960807
Iteration 19000 , Time per 1000 iterations: 1.006917
Iteration 20000 , Time per 1000 iterations: 1.058458

## Alternative execution
Iteration 1000 , Time per 1000 iterations: 0.022474
Iteration 2000 , Time per 1000 iterations: 0.022412
Iteration 3000 , Time per 1000 iterations: 0.022726
Iteration 4000 , Time per 1000 iterations: 0.022411
Iteration 5000 , Time per 1000 iterations: 0.023426
Iteration 6000 , Time per 1000 iterations: 0.022177
Iteration 7000 , Time per 1000 iterations: 0.0224
Iteration 8000 , Time per 1000 iterations: 0.022154
Iteration 9000 , Time per 1000 iterations: 0.024215
Iteration 10000 , Time per 1000 iterations: 0.022154
Iteration 11000 , Time per 1000 iterations: 0.021894
Iteration 12000 , Time per 1000 iterations: 0.022275
Iteration 13000 , Time per 1000 iterations: 0.022152
Iteration 14000 , Time per 1000 iterations: 0.022036
Iteration 15000 , Time per 1000 iterations: 0.022192
Iteration 16000 , Time per 1000 iterations: 0.022265
Iteration 17000 , Time per 1000 iterations: 0.026411
Iteration 18000 , Time per 1000 iterations: 0.021993
Iteration 19000 , Time per 1000 iterations: 0.022269
Iteration 20000 , Time per 1000 iterations: 0.022074
```

klayoutmatthias, 2025-07-16:
Hi @engineeringvase,

thanks for bringing this up.

It's in there by design, but I guess it can be modified.

First of all, there is a simple workaround: use "start_changes" and "end_changes" on the Layout:

```
try:

  ly.start_changes()

  print("")
  print("## Problematic case")
  t0 = datetime.datetime.now()
  for i in range(1,20001):
      
      # Time keeping for demo
      if i%1000 == 0: 
          timecost = (datetime.datetime.now() - t0).total_seconds()
          print("Iteration", i, ", Time per 1000 iterations:", timecost)
          t0 = datetime.datetime.now() 
          
      # Problematic execution
      main.shapes(l1).insert(box1)
      main.shapes(l1).insert(box2)
      
finally:
  ly.end_changes()
```

this will bring the runtimes down in the first case.

The reason is that Region is not just a container for shapes, it has other faces. For example, it can represent a hierarchical layer in a layout. If you insert that one into a Shapes container, it will flatten this layer and insert the flat shapes into the Shapes container. 

As a special (but not unusual) case, the Region may refer to a layer from the same Layout object than the one the Shapes container lives in. This needs special care as recursively iterating the source layer needs a properly built cell tree, region lookup trees and other auxiliary structures. In that case, there is an interaction between reading and writing (which invalidates these structures) with potentially harmful side effects.

The mitigation strategy is to lock the layout's lookup structures during the insert process, but that comes with a rebuild of the lookup structures at the beginning of every insert. As this takes O(n) time (n = number of shapes), you get the increasing time adder. Adding another lock around the sequence will prevent that. The "insert" method alone can't do that as it does not know when you're finished.

Having said that, I think I can remove the internal locking if the copy source is a separate Shapes object or a different Layout. But not in all cases.

Matthias

klayoutmatthias, 2025-07-17:
Hi @engineeringvase,

Depending on the KLayout version you use, the Box objects you add may be of a different kind when you add them through iteration (Polygon with properties) or not (Polygon). I have not checked the details yet, but if that is the case, these variants form different slots and O(n) applies to every slot individually. Hence the slow variant continues where it ended.

Matthias