KLayout/klayout #2100: RNetExtractor extracts wrong RNode locations with Algorithm.Tesselation

martinjankoehler, 2025-07-29:
Hey @klayoutmatthias,

I noticed an issue with the extracted node locations.
`SquareCounting` gives me the correct locations, but with `Tesselation` the location get's messed up.

Script to reproduce:
```python
from typing import *
import klayout.db as kdb
import klayout.pex as klp

#--------------------------------------------------------
def run_test(algorithm: klp.Algorithm) -> klp.RNetwork:
    tech = klp.RExtractorTech()

    M1 = 10
    M2 = 20
    V1 = 15

    for l in (M1, M2):
        c = klp.RExtractorTechConductor()
        c.layer = l
        c.resistance = 1000.0
        c.algorithm = algorithm
        tech.add_conductor(c)

    v = klp.RExtractorTechVia()
    v.bottom_conductor = M1
    v.cut_layer = V1
    v.top_conductor = M2
    v.resistance = 5000.0
    tech.add_via(v)
    #--------------------------------------------------------

    polygon1 = kdb.Polygon([kdb.Point(340,235), kdb.Point(340,885), kdb.Point(600,885), kdb.Point(600,235)])

    geo: Dict[int, kdb.Region] = {
       M1: kdb.Region([polygon1]),
       V1: kdb.Region([polygon1]),
       M2: kdb.Region([polygon1]),
    }
    vertex_ports: Dict[int, List[kdb.Point]] = {}
    polygon_ports: Dict[int, List[kdb.Polygon]] = {
       M1: [polygon1]
    }

    #--------------------------------------------------------
    rexer = klp.RNetExtractor(dbu=0.001)
    nw = rexer.extract(tech, geo, vertex_ports, polygon_ports)
    return nw

for a in (klp.Algorithm.SquareCounting, klp.Algorithm.Tesselation):
    nw = run_test(a)
    print(f"\n{a} Network with {nw.num_nodes()} nodes, {nw.num_elements()} elements")
    for n in nw.each_node():
        print(f"\tNode id={n.object_id()}, layer={n.layer()}, location={n.location()}")
```

Output I get:
```
SquareCounting Network with 1 nodes, 0 elements
	Node id=105553713496352, layer=10, location=(0.34,0.235;0.6,0.885)     # CORRECT

Tesselation Network with 1 nodes, 0 elements
	Node id=105553713496928, layer=10, location=(-0.13,-0.325;0.13,0.325)    # WRONG
```

klayoutmatthias, 2025-07-29:
Hi @martinjankoehler,

thanks for that perfectly prepared test case.

Looks like I left a debug statement ("Graph written to debug.gds") too :(

I'll try to debug and fix the problem.

Best regards,

Matthias

martinjankoehler, 2025-07-29:
Awesome @klayoutmatthias, thanks for the quick fix!

closed 2025-07-30 by klayoutmatthias