KLayout/klayout #2107: Loading large GDS files with slow performance

zhangtaohbwh, 2025-08-06:
Hello, could I ask a question? I have developed a GDS file editor based on the C++ Qt image view framework (using QGraphicsView, QGraphicsScene, and QGraphicsItem). When dealing with a very large GDS file (over 200MB) containing over 2 million polygon primitives, the rendering process in QGraphicsView becomes extremely slow after loading all these primitives into QGraphicsScene. Are there any solutions or reference cases you could suggest to address this issue?

ejprinz, 2025-08-06:
You shouldn't be using GDSII for large design databases, this format has
been obsolete for 10's of years. Use OASIS instead, the files are 10-100x
smaller compared to uncompressed GSDII.

stefanottili, 2025-08-06:
@ejprinz 

a) gds is not obsolete, it's alive and kicking
reason being that oasis reader/writer that take advantage of the format are a lot more complex to write and
due to that also more prone to bugs than gds reader/writer.

b) yes, oasis is the default for large chips or mask data
large being couple hundred MB gds.gz, anything smaller reads/writes in seconds, there not much to gain using oasis.

10-100x smaller oasis is in reality closer to 10'isch then 100 ...

c) one should be using gzipped gds (zlib level 1 reads/writes 2x faster then the default 6 => 10% larger files)
typically you get ~80% compression ratio.
In a datacenter environment, smaller files transfer faster.

d) one should be using strict, cblock (zlib compression) oasis
oasis cblocks only achieve ~30% compression ratio due oasis already having run-length encoded data elements.

if c) and d) are met, then a 600MB (3.5GB) gds.gz corresponds to a 150MB (230MB) oasis, that's a 4x advantage for oasis.
the 15x difference in uncompressed data doesn't translate in oasis reading 15x faster.

e) strict oasis allows to implement a multithreaded oasis reader, this can speed up reading a lot,
but it depends heavily on the file-io speed. Much faster from a local ssd than from a datacenter nas.

f) rendering speed is (mostly) independent of the input format.
klayout -ne takes advantage of oasis repetitions 10/11 to have a smaller memory footprint, that speeds up drawing a little bit.

closed 2025-08-11 by zhangtaohbwh