KLayout/klayout #2108: Build failure on macOS Sequoia: Undefined symbol gsi::cls_decl<db::Net>() in dbMeasureEval.o

Kazzz-S, 2025-08-06:
Hello @klayoutmatthias,

First of all, thank you for your efforts in preparing the 0.30.3 release.

**Description of the issue**
Building the current `master` branch on macOS Sequoia (Intel) fails with an undefined symbol error related to `db::Net` when linking `libklayout_db`.

**Environment:**
- macOS Sequoia 15.6 (Intel)
- Apple clang version 17.0.0 (clang-1700.0.13.5)
- Target: x86_64-apple-darwin24.6.0
- KLayout commit: 10a34cca (master branch, 0.30.3-to-be)
  (built locally with a slight modification: ef365b55 to bypass the link error)
- Build system: `build4mac.py` with the default options.

**Error log:**
```
Undefined symbols for architecture x86_64:
"gsi::ClassBase const* gsi::cls_decldb::Net()", referenced from:
db::MeasureNetEval::net_func() const in dbMeasureEval.o
ld: symbol(s) not found for architecture x86_64
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
```

**Analysis (by ChatGPT):**
The function `MeasureNetEval::net_func()` (in `src/db/dbMeasureEval.cc`) calls:
```
return tl::Variant::make_variant_ref(n->second);
```
where n->second is a const db::Net*.
This requires a GSI binding for db::Net (gsi::cls_decl<db::Net>()).
I confirmed that such a binding does not exist in the current source tree by searching (`grep -R "cls_decl<db::Net>" src/`), which returned no results.

This causes a link error on macOS (and possibly other platforms with stricter linking).
It may have been macOS’s stricter linker that surfaced this issue.

**Workaround (by ChatGPT):**
As a temporary fix, replacing the line (Left) with a dummy `return tl::Variant();` (Right) allows the build to complete.
[image]

**Rebuild**
Succeeded 😄 
[image]

Best regards,
Kazzz-S

klayoutmatthias, 2025-08-06:
Hi @Kazzz-S,

thanks for reporting this issue. It looks like a way to reproduce it are the Python module builds. I will try to fix it.

Best regards,

Matthias

stefanottili, 2025-08-06:
builds on M1 homebrew

klayoutmatthias, 2025-08-06:
Thanks ... you mean you tried the patch?

ARM is still running, but the other Python actions are good now :)

Matthias

stefanottili, 2025-08-06:
Master with 2108 builds on Mac.

Kazzz-S, 2025-08-07:
The `master` branch for issue #2108 builds on my Intel Mac, too.
Thank you for the quick fix.

Kazzz-S