KLayout/klayout #2127: Offgrid shapes created when saving gds file as cif.

d-m-bailey, 2025-08-20:
Hi @klayoutmatthias 

First of all, kudos for klayout and it's ease of use.

Using klayout 0.30.1 r65a9465bc on macos.

When I open [this gds layout](https://github.com/IHP-GmbH/TO_Apr2025/blob/main/160GHz_LNA/design_data/gds/FMD_QNC_04_160GHz_LNA.gds) and save it as a `cif` file, I notice that some points are saved as off grid.

Here's the original shape.

[image]

Immediately after saving with these options,

[image]

you can see that there is no change.

[image]

But when I read the file back in, the lower-left y-coordinate has shifted by 0.001 from -0.785 to -0.786 which is no longer on the 0.005um grid.

[image]

I'm investigating the use of `cif` files (ascii text) instead of binary format so that small changes in the design can easily be tracked with tools like git.

CC: @H-Ojiro

stefanottili, 2025-08-20:
It looks to me as if the cif box format is 'B width height center_x, center_y'
```
B 700 1555 5550,-8;
```
You need to have your width/height to be a multiple of 10, not 5 to avoid this kind of rounding error.

stefanottili, 2025-08-20:
if you want to save diskspace and have fast read/write times, use oasis instead of gds.

d-m-bailey, 2025-08-20:
@klayoutmatthias @stefanottili 

Thanks for your comments.

I'm working on a [jupyter notebook](https://github.com/d-m-bailey/ihp-mpw-be/blob/main/TO_202504.ipynb) that will run LVS (magic/netgen & klayout) along with my CVC-RV (ERC checks) on the iHP open source shuttle data without having to install anything locally. 

I was looking for a way to track the (non-mask) changes that need to be made to the layout to pass LVS.

Rather than store the entire modified layout file, I think I can just record a macro in klayout to modify the layout and then track that.

Thanks again for your prompt response.

d-m-bailey, 2025-08-21:
@klayoutmatthias quick question.

Is there a may to record gui edits in klayout, save them to a file, and replay them later similar to how Excel allows you to record actions in a macro?