KLayout/klayout #2133: Crash in Macro Development: when using find and replace

martinjankoehler, 2025-08-31:
## Steps to reproduce

- open *Macro Development*
- open attached script (needs to be at least so long that the scrollbar is involved)
- enter find query: `if DEBUG`
- use the > arrow to get replacement text field
- enter replacement: `if debugging.DEBUG`
- hit replace ALL button
- EXPECTED: replaced strings
- OBTAINED: crash

```
ERROR: Signal number: 11
Address: 0x0
Program Version: KLayout 0.30.3 (2025-08-08 re359f2af2)

Backtrace:
/Users/martin/Source/klayout/qt5MP.build.macos-Sequoia-release-Rhb34Phbauto.macQAT/libklayout_lay.0.30.3.dylib +0x1e7178 _ZN3lay14signal_handlerEiP9__siginfoPv
/usr/lib/system/libsystem_platform.dylib +0x36a4 _sigtramp
/opt/local/libexec/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets +0x121728 _ZN14QPlainTextEdit13updateRequestERK5QRecti
/opt/local/libexec/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets +0x121728 _ZN14QPlainTextEdit13updateRequestERK5QRecti
/opt/local/libexec/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets +0x122048 _ZNK14QPlainTextEdit17firstVisibleBlockEv
/opt/local/libexec/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets +0x1252cc _ZN24QPlainTextDocumentLayout11qt_metacallEN11QMetaObject4CallEiPPv
/opt/local/libexec/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets +0x13d05c _ZN18QWidgetTextControl13setTextCursorERK11QTextCursorb
/Users/martin/Source/klayout/qt5MP.build.macos-Sequoia-release-Rhb34Phbauto.macQAT/libklayout_lay.0.30.3.dylib +0xe42bc _ZN3lay15MacroEditorPage20replace_in_selectionERK7QStringb
/Users/martin/Source/klayout/qt5MP.build.macos-Sequoia-release-Rhb34Phbauto.macQAT/libklayout_lay.0.30.3.dylib +0xcb490 _ZN3lay17MacroEditorDialog26replace_all_button_clickedEv
/opt/local/libexec/qt5/lib/QtCore.framework/Versions/5/QtCore +0x149138 _ZN11QMetaObject8activateEP7QObjectPKS_iPPv
/opt/local/libexec/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets +0x9c2bc _ZN15QAbstractButton7clickedEb
```

[image]

martinjankoehler, 2025-08-31:
@klayoutmatthias good point ;-)
Added in the original post.

klayoutmatthias, 2025-08-31:
Reproduces on Linux ... very good. Thanks!

closed 2025-09-06 by klayoutmatthias