KLayout/klayout #2138: flag_missing_ports flags all internal nets for source subcircuits with no ports

d-m-bailey, 2025-09-03:
Requires at least klayout 0.30.2 (`flag_missing_port` helper routine).

@klayoutmatthias Thanks for the port check enhancement. That has helped find discrepancies in standard cell port labeling.

However, we noticed that when the source spice top cell has no ports, then all of the internal nets are considered ports and cause a mismatch.

[image]

You can verify this with the `IHP-Open-PDK` repo test cases.

```
git clone https://github.com/H-Ojiro/IHP-Open-PDK.git --depth=1 -b port_check
cd IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/tech/lvs/testing
python3 run_regression.py RES
```

Here's the spice 
```
.SUBCKT rppd
Rp1 net1 net2 rppd m=1 l=0.5u w=0.5u
Rp2 net3 net4 rppd m=1 l=1u w=0.5u
Rp3 net5 net6 rppd m=1 l=1.2u w=0.7u
Rp4 net7 net8 rppd m=1 l=1.5u w=0.5u ps=0.2u b=1
Rp5 net9 net10 rppd m=1 l=1.7u w=0.5u ps=0.2u b=2
...
```

The layout has no text.

CC: @H-Ojiro

klayoutmatthias, 2025-09-03:
Problem is that inside your LVS script, you do a "simplify" on the netlists (optional, but selected here). This implies "make_top_level_pins" (see https://www.klayout.de/doc-qt5/code/class_Netlist.html#m_simplify). Hence, the schematic has top-level pins which are not found in the layout. This causes these errors.

You should skip the simplify step as it is redundant with the other netlist manipulation methods ([make_top_level_pins](https://www.klayout.de/doc-qt5/code/class_Netlist.html#m_make_top_level_pins), [purge](https://www.klayout.de/doc-qt5/code/class_Netlist.html#m_purge), [combine_devices](https://www.klayout.de/doc-qt5/code/class_Netlist.html#m_combine_devices) and [purge_nets](https://www.klayout.de/doc-qt5/code/class_Netlist.html#m_purge_nets).

Matthias

d-m-bailey, 2025-09-04:
@klayoutmatthias Thanks for the quick response.

I see that `simplify` is set as the default and understand that this implies `make_top_level_pins`.

Maybe my question is about the functionality of `make_top_level_pins`. On the schematic side, I assumed that it marked the top level subckt **ports** as pins and on the layout side marked the top level **labeled nets** as ports.

However, if there are no subckt ports like this
```
.SUBCKT rppd
Rp1 net1 net2 rppd m=1 l=0.5u w=0.5u
Rp2 net3 net4 rppd m=1 l=1u w=0.5u
Rp3 net5 net6 rppd m=1 l=1.2u w=0.7u
Rp4 net7 net8 rppd m=1 l=1.5u w=0.5u ps=0.2u b=1
Rp5 net9 net10 rppd m=1 l=1.7u w=0.5u ps=0.2u b=2
...
```
then all the internal nets at the top level of the schematic are considered ports as shown here.

[image]

If I add even one port to the subckt like this
```
.SUBCKT rppd a_1
Rp1 net1 net2 rppd m=1 l=0.5u w=0.5u
Rp2 net3 net4 rppd m=1 l=1u w=0.5u
Rp3 net5 net6 rppd m=1 l=1.2u w=0.7u
Rp4 net7 net8 rppd m=1 l=1.5u w=0.5u ps=0.2u b=1
Rp5 net9 net10 rppd m=1 l=1.7u w=0.5u ps=0.2u b=2
...
```
The mismatch result is only the `a_1` net.

[image]

klayoutmatthias, 2025-09-04:
Yes, it is. But the way the IHP LVS uses it is not intended.

The whole set of "purge*", "make_top_level_pins" and "combine_devices" features is intended to be used on **Layout** netlists.

"purge" and "purge_devices" is needed to eliminate dummy devices and floating nets arising from dummy fill pattern for example. "combine_devices" is needed to combine serial resistors into a single electrical device for example. "make_top_level_pins" is required to provide pins for pin matching. The idea is that schematic netlists follow certain conventions - e.g. they may or may not include dummy devices or they represent a multi-part resistor chain by a single device. Depending on what your conventions are, you can choose to use "purge_devices" or "combine_devices" on the Layout netlist, so it matches the schematic one. The intention was not to use it one both layout and schematic. That would eventually dilute the information of the schematic - so when the schematic includes dummy devices, there is a reason and they should be checked too. Or if a serial resistor chain is represented by multiple resistors, they should be checked against a serial chain in the layout. So you would choose not to call "combine_devices".

On top level, pins are represented by the pads. Lacking any further definition of a pad, the first approach is to treat every labelled net as one that connects to a pad. Often the pad frame is a overlay cell at top level and you can place labels on the pads there without any other labels in that cell. Using "make_top_level_pins" on the layout netlist will then give you pins for the pads. Hence the definition of "make_top_level_pins" currently is to create pins for all labelled nets. As the netlist API is expose, you should be able to code other schemes as well. 

My general feeling is that KLayout is expected to digest any kind of schematic/layout style combination. This is simply not possible. Some conventions need to be established. If someone has a good suggestion how these conventions may look like, I am open to a discussion. 

Matthias

d-m-bailey, 2025-09-04:
@klayoutmatthias Thanks again for a detailed and easy to understand explanation. 

I understand now that the `simplify` routines were not intended to be used on the schematic netlists. I will suggest the changes to IHP.