KLayout/klayout #2149: Marker database error

KrzysztofHerman, 2025-09-09:
After running the DRC using IHP new deck the marker browser does not open complaining about the following issue:

`ERROR: XML parser error: bondpad_70x70:r90 is not a valid cell name or name/variant combination in line 8787, column 33  `

This is the respective piece of the *.lyrdb file

```
  <item>
   <tags />
   <category>'Pad.d1R'</category>
   <cell>bondpad_70x70:r90</cell>
   <visited>false</visited>
   <multiplicity>1</multiplicity>
   <comment />
   <image />
   <values>
    <value>edge-pair: (2.135,67.86;67.86,67.86)/(75.553,76;-5.558,76)</value>
   </values>
  </item>
  <item>

```

The cell itself appears in the layout hierarchy as follows:

[image]

klayoutmatthias, 2025-09-09:
BTW: what KLayout version are you using?

Matthias

KrzysztofHerman, 2025-09-10:
Hi, @klayoutmatthias 

Here goes the database:

[FMD_QNC_Padframe24_FMD_QNC_Padframe24_full.lyrdb.gz](https://github.com/user-attachments/files/22247913/FMD_QNC_Padframe24_FMD_QNC_Padframe24_full.lyrdb.gz)

It was generated using `Klayout 0.30.2` and new DRC deck from the `dev` branch.

The design itself sits [here](https://github.com/KrzysztofHerman/Open-Padrings/blob/main/24/final/gds/FMD_QNC_Padframe24.gds)

BTW I have sent you some mails last week but I got mail delivery issue.

klayoutmatthias, 2025-09-10:
Very good. Thanks. I will have a look.

I was forcible migrated to Microsoft 365 mail service by my hoster (thanks GoDaddy c/o DomainFactory!!!) which created some major issues on my side - like loosing one year of mails which they failed to migrate. You may also encounter problems sending mails to me because Microsoft seems to intentionally blacklist other providers. I guess it's their way to force everyone on their platform. Please try again, but if it still does not work, all I can do as of now is to complain to DomainFactory - not for the first time and likely with the same effect as before, which is zero. I don't have time currently to set up a dedicated mail service and move the domain. I'm sorry. 

Matthias

klayoutmatthias, 2025-09-10:
Hmm ... the only mail that arrived was from GitHub notifying me about your comment. That is bad :( I will check my junk mail later. Maybe it went there.

Anyway, I need your help: I can reproduce the problem with the DRC wrapper script. But when I try to break down the problem and run KLayout directly, I receive a nice and clean RDB file. The way I run DRC directly from KLayout is this:

```
klayout -rd input=FMD_QNC_Padframe24.gds -rd report=x.lyrdb -rd log=run_drc.log -rd thr=12 -rd run_mode=deep -b -r IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/tech/drc/rule_decks/sg13g2_maximal.drc
```

I don't fully understand the things the wrapper is doing. Maybe there is some RDB manipulation ongoing which fails?

Thanks,

Matthias