KLayout/klayout #2152: edit_menu.mode_menu.* shortcuts for the toolbar have no effect (on latest Archlinux KLayout package)

bobo38, 2025-09-20:
[image]

I've been trying to edit some layout on IHP 0.13um PDK with the KLayout 29.9 on Archlinux with my ShortCut setup (see the setup snapshot). It seems that all my "edit_mode.menu_mode" shortcuts are not working: if I press the keys, it does nothing, as if the shortcut was not declared. To my knowledge all my other shortcuts are working fine and produce the correct effect.

Trying to downgrade the Klayout package to "klayout-0.29.9-1-x86_64.pkg.tar.zst" from "klayout-0.29.9-3-x86_64.pkg.tar.zst" results in some error:
`klayout: error while loading shared libraries: libruby.so.3.3: cannot open shared object file: No such file or directory`
I do not have other KLayout package in my cache, so that it is not easy to go back to a version with a working setup. I didn't have this issue previously.

If you need more information, I could provide them. It's possibly an Archlinux packaging issue. If you cannot reproduce the issue on your side, I will try to contact the Archlinux packager.

Thank you for all the great work with KLayout, it's an amazing piece of software

bobo38, 2025-09-20:
Reading the release notes there is bug fix from 0.29.10 that looks like related to this bug. Once I will be able to run my own compiled from source KLayout, this issue will be most likely fixed.

bobo38, 2025-09-20:
working on compiled 0.30.4 on Archlinux :-) Sorry for reporting outdated bug

klayoutmatthias, 2025-09-21:
Very good. Thanks for letting me know.

But you could have simply closed the issue :)

Matthias

closed 2025-09-21 by klayoutmatthias